lowRISC/ibex #1978: Tested FPGA Platforms for Ibex

parsnip908, 2023-01-30:
Hello,
My name is Pranav Kharche and I am a senior at UC Davis. My senior design team would like to use the Ibex core as a base for our project (customizing a RISC-V core). We have some FPGAs available but would like to know if the core was ever tested or used on these platforms. 
* Zedboard - a development board for Xilinx Zynq-7000 series FPGA. https://www.xilinx.com/products/boards-and-kits/1-8dyf-11.html
* Nexys A7 - Xilinx Artix-7 FPGA https://www.xilinx.com/products/boards-and-kits/1-6olhwl.html
* DE1-SOC - Altera Cyclone V FPGA http://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=167&No=836&PartNo=4

## My Environment

**EDA tool and version:**
Vivado Standard edition 2022.2
or Quartus Prime standard 20.2
or whatever you suggest

**Operating system:**
Windows 10

**Version of the Ibex source code:**
Latest version. No changes made.

hcallahan-lowrisc, 2023-01-30:
Hi Pranav! Thanks for your interest in the Ibex core. I hope I can help you out.

I don't believe we have any examples of Ibex synthesized on the platforms you have mentioned. However there is no reason it couldn't be, and with a bit of effort it shouldn't be too difficult to create a top level that wraps the Ibex core for any of those boards. Perhaps the easiest option would be to start with the [ibex-demo-system](https://github.com/lowRISC/ibex-demo-system) repository, which contains a simple ibex + peripherals system there that runs on the [Arty-A7](https://digilent.com/shop/arty-a7-artix-7-fpga-development-board/) board (an Artix-7 FPGA), which is a very similar FPGA to the Nexys A7 board you have. Adapting the demo system for the Zedboard (a Zynq-7000) should also be possible, utilizing just the FPGA fabric without the hard PS. Adapting for an Intel/Altera system should also be possible but would require a little more work (e.g. to map the RISCV debug module onto the user JTAG TAP).
We recently used this demo system as part of a lab at the [HiPEAC 2023 conference](https://www.hipeac.net/2023/toulouse/#/program/sessions/8019/), and there are accompanying lab notes [here](https://github.com/lowRISC/ibex-demo-system-labs) which you may find useful getting started.

Unfortunately we don't currently use Windows at all for development, so adapting the scripting and tutorials we have documented may prove a challenge. If you are confident working in a WSL or cygwin/minGW environment you might be able to get it working, but working from linux (e.g Ubuntu) would be much easier. We're sorry that we can't support Windows users better at this time, it's an area we would like to improve in. There are some windows-specific commands in the repositories we tried to use for the lab participants, but we're not super-confident in them.

I hope this was helpful. We also have a [Zulip channel](https://lowrisc.zulipchat.com/#narrow/stream/198227-ibex) for chatting about Ibex too if you have any more questions!

Thanks,
Harry

hcallahan-lowrisc, 2023-01-30:
I just did a quick scan through some of the [older chats](https://lowrisc.zulipchat.com/#narrow/stream/198227-ibex/topic/zedboard.20.26.20genesys2.20examples/near/279394860) on our Zulip channel and found [this repository](https://github.com/mbaykenar/ibex_fpga) extending the ibex fpga examples to support the NexysA7 and Zedboard platforms! I haven't looked too closely but you may find this a good starting point if you choose to pursue it.

GregAC, 2023-01-31:
I'd note you're better off with Xilinx FPGAs. I have actually synthesised and used an Ibex core on an Intel Cyclone V but their software doesn't support many of the system verilog constructs we use so you need to use something like https://github.com/zachjs/sv2v to convert the system verilog to verilog first and then synthesise that. Overall it's simpler to use the Xilinx tool flow.

rswarbrick, 2023-05-22:
Hi there! Closing this issue, because I don't *think* there's anything that we need to do for it. Please feel free to re-open if there's something I've missed. Thanks!

closed 2023-05-22 by rswarbrick