lowRISC/ibex #1987: Ephemeral Simulated Environments for Testing Ibex Core

ShrutiC-git, 2023-02-20:
Hi, Ibex community!

Our team at Uffizzi met some of the colleagues who work on Open Titan at OSS, Dublin, and discussed possible utility for ephemeral simulated test environments for embedded systems. We had developed a Proof of Concept for OpenTitan, and wanted to move further into exploring ephemeral, shareable, simulated environments for testing Ibex-core. We have developed a Proof of Concept for Ibex and would like to share it with the community ~

In this [fork of Ibex-core](https://github.com/ShrutiC-git/ibex/pull/2), we set up a GHA workflow that uses Uffizzi to create, update, and delete an ephemeral simulated test environment on every PR. The environment is built using a `Dockerfile` which was added in the PR and comes with the needed dependencies to build and test Ibex. The environment will get started with `Verilator`, `RISCV-toolchain`, `gcc`, `llvm` and other tools needed to build the simulated hardware and run the software on top of it. 

An example environment is running on [Uffizzi Cloud](https://uffizzi.com/) and is accessible via https [here](https://pr-2-deployment-16751-ibex.app.uffizzi.com/).

We ran the `simple_system` test on the simulated core, and the test passed successfully. Attached is a screenshot of the test passing on the ephemeral environment created for the PoC. 
[image]

The environment will be up if you want to try it yourself. The entrypoint is `Ibex-repo-base`. The `home` directory contains other tools needed for running the simulation. 

Please let us know if this would be useful to Ibex and in general if it's useful to embedded systems. We can also discuss running the testbench UVM in the ephemeral environment which requires commercial RTL simulators.
 
  @marnovandermaas
  @jpthurman

marnovandermaas, 2023-02-28:
Thanks for this work, we use the Spike ISA simulator in conjunction with RISCV-DV and UVM to verify our Ibex core. Am I right in assuming this environment just has the ISA simulator in it? I hope you found using our open source project a useful way of testing your product.

ShrutiC-git, 2023-02-28:
Hi, @marnovandermaas! 

Thanks for writing back, and explaining the process of testing the core in more detail. Ibex was definitely a very cool project in helping us explore embedded systems.

Yes, that's right - this current iteration only has the Spike ISA Simulator (the RISC-V ISA simulator). While trying to run the testbench against Ibex core, the documentation mentioned a couple of prereqs, mainly a Spike ISA simulator and a SystemVerilog simulator with UVM support, so I didn't install it explicitly in the PR.

However, To get Spike to run the testbench, we did try to install riscv-dv along with riscv-pk. But that did not work because this current iteration does not have the Verilog simulator with UVM support. I checked the simulators currently supported by Ibex core in the `simulator.yaml` script and found that these were commercial simulators. So far we looked and found "in-development" versions of Verilator with UVM support, but the support for Verilator might not have yet been added to Ibex for that reason.

Maybe there's a workaround this I might have missed. Would you be able to point to what/how we can meet the requirements for the test bench?

marnovandermaas, 2023-03-22:
Hi @ShrutiC-git, thanks for this explanation. Yes, our current testing environment requires UVM and as far as I know Verilator's UVM support is in development. It would be great to have an open source tool support UVM, but at the moment the UVM part of the repo requires a commercial simulator.

rswarbrick, 2023-05-22:
Hi there! Thanks for the messages above, and also thanks for explaining what you were doing on this blog post: https://www.uffizzi.com/blog/simulated-ephemeral-cloud-environments-for-embedded-systems.

It looks very cool, and I hope that the up-and-coming free software simulation tools will be able to do stuff for us soon: I can't wait either!

I'm going to close this issue now, since I don't *think* there's really anything we can do right now to improve things. Please re-open it or open new issue if I've missed something. Many thanks.