lowRISC/ibex #1994: Using UVM environment for post-synthesis and post-layout simulation

LorenzoLagostina, 2023-03-05:
Hello,
my name is Lorenzo Lagostina, I'm working on a research project in the HLS lab at the Department of Electronics and Telecommunications of the Polytechnic of Turin, together with PhD student Filippo Minnella and professor Luciano Lavagno. 

The research project is focused on back-end techniques, and we decided to apply them to your ibex processor (btw congratulations on your excellent work with it :D). We made this choice both to assess the advantages of synthesis and layout of a processor, and to verify the correct functionality of the design using your UVM environment.

Therefore, we want to use your verification environment on the post-layout netlist, in order to assure the correct functionality of the ibex core after we applied our back-end techniques (I cannot give detailed information on such techniques since they are undergoing patent review).

We followed the setup instructions and successfully executed the ibex simulations for your RTL design, resulting in the expected coverage.
We are now executing the physical synthesis of the ibex core, and want to use the UVM environment on the post-synthesis (for starting). 
What we did is the following:
- Synthesise the ibex_core entity (together with all the sub-entities), and obtained the verilog netlist (flattened) and a system verilog wrapper (changing the wrapper's name to ibex_core and the netlist one to ibex_core_netlist).
- Added these two files to the ibex_dv.f file (and commented the original ibex_core.sv) to include them in the simulation.
- Launched the UVM environment with the following options: 

> make IBEX_CONFIG=small SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test COSIM=1 VERBOSE=1 COSIM=1 --debug

The simulation prints the warnings and errors in the out/build/tb/compile_tb_stdstreams.log file:

> Warning-[UII-L] Interface not instantiated
/home/lorenzo/ibex_last_update/ibex/dv/uvm/core_ibex/fcov/core_ibex_fcov_if.sv, 8
core_ibex_fcov_if
  Interface 'core_ibex_fcov_if' defined in logic library 'work' is never
  instantiated in design. It will be ignored.
> Warning-[UII-L] Interface not instantiated
/home/lorenzo/ibex_last_update/ibex/dv/uvm/core_ibex/fcov/core_ibex_pmp_fcov_if.sv, 7
core_ibex_pmp_fcov_if
  Interface 'core_ibex_pmp_fcov_if' defined in logic library 'work' is never
  instantiated in design. It will be ignored.
> Error-[SVA-COBMMI] Can only bind to modules or instances.
/home/lorenzo/ibex_last_update/ibex/dv/uvm/core_ibex/fcov/core_ibex_fcov_bind.sv, 14
core_ibex_fcov_bind
  Can only bind to modules or module instances.
  bind ibex_core core_ibex_pmp_fcov_if #(.PMPGranularity(PMPGranularity),
  .PMPNumRegions(PMPNumRegions), .PMPEnable(PMPEnable)) u_pmp_fcov_bind(.*); 
> Error-[SVA-COBMMI] Can only bind to modules or instances.
/home/lorenzo/ibex_last_update/ibex/dv/uvm/core_ibex/fcov/core_ibex_fcov_bind.sv, 6
core_ibex_fcov_bind
  Can only bind to modules or module instances.
  bind ibex_core core_ibex_fcov_if u_fcov_bind(.*);

Our assumption is that the synthesizer discards verification-related signals present in the RTL file. Therefore, since the simulator cannot find such signals, it discards the core_ibex_fcov_if and core_ibex_pmp_fcov_if interfaces.
We are now going to try to set all verification-related signals in the design as "dont_touch" for the synthesizer, hoping it will solve our problem.

In your opinion, is there something that we missed that might cause this issue or some advice you could give us? Did you use the UVM environment on post-synthesis and/or post-layout netlists? 

Thanks in advance for your attention and for any advice you might give us.
Kind regards,
Lorenzo Lagostina
## My Environment

**EDA tool and version:**
I'm using VCS 22.06 for simulations, Design Compiler 20.09 for synthesis, and Innovus 20.11 for P&R.

**Operating system:**
CentOS 8

**Version of the Ibex source code:**
5693d7da3264f96e52b05a496cf447fe532606f7

GregAC, 2023-03-06:
We've not run DV on post synthesis/post layout netlists.

The thing that's failing is binding in the interfaces that provide functional coverage. Personally I just wouldn't bother with the functional coverage collection. You're applying DV to the post synthesis netlist to check it's functional, you don't close coverage on it so there's no need to have it. If you just remove those interface files and the `core_ibex_fcov_bind.sv` file from the filelist used for DV that should fix the issue.

There may be other points we probe from UVM that could cause issues but the functional coverage collection is by far the most invasive.

LorenzoLagostina, 2023-03-07:
Got it, we'll try that then.

Thank you very much for the advice and the quick response!