lowRISC/ibex #2011: Ask a question related to Ibex

raiyyanfaisal09, 2023-04-06:
I am trying to run the verification for the core. I have installed the toolchain as instructed. However, while trying to install and build spike simulator. I am seeing these issues. I have gone through the same steps that have been mentioned on the README for https://github.com/lowRISC/riscv-isa-sim  

[image]

the command I used to configure the tool was 

../configure --enable-commitlog --enable-misaligned --prefix=/home/rmasumda/projects/spike_cosim

and then I ran make  and that is when I encountered that error.

## My Environment

                                       
I am using 15fbd5680e44da699f828c67db15345822a47ef6 version of the IBEX. I am using 2017.12-1 version of Synopsys VCS, 3.900 version of Verilator, 7.2.0 version of GCC and 3.7.3 version of python

The Linux is CentOS, version 7. 

Can you please tell me what the problem is and how can it be fixed?

rswarbrick, 2023-04-11:
Hi there,

I'm sorry, but I can't actually find that SHA in either Ibex or our spike clone (riscv-isa-sim). Please could you reply with a description of what exactly you were building?

Thanks!

raiyyanfaisal09, 2023-04-18:
@rswarbrick, that did work! thanks! However, I could still see the other error I mentioned above. There are some `defines used and I cannot see them being defined anywhere. Is there a file missing ?

Error-[UM] Undefined macro
/home/rmasumda/projects/ibex_riscv/dv/uvm/core_ibex/env/core_ibex_dut_probe_if.sv, 92
  Undefined macro exists as: 'DV_CREATE_SIGNAL_PROBE_FUNCTION'   
  "/home/rmasumda/projects/ibex_riscv/dv/uvm/core_ibex/env/core_ibex_dut_probe_if.sv",
  92: token is '`DV_CREATE_SIGNAL_PROBE_FUNCTION'
    `DV_CREATE_SIGNAL_PROBE_FUNCTION(signal_probe_rf_ren_a, rf_ren_a)

raiyyanfaisal09, 2023-04-18:
I also tried to run the sims with Xcelium and its complaining about a totally different file. 

make[1]: *** [out/metadata/tb.compile.stamp] Error 1
Generating core configuration file
Building randomized test generator
Running randomized test generator to create assembly file /home/rmasumda/projects/ibex_riscv/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/test.S
Compiling riscvdv test assembly to create binary at /home/rmasumda/projects/ibex_riscv/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/test.bin
make[1]: Target `all' not remade because of errors.
make: *** [run] Error 2

& the following is the error from compile_tb.log 

    if (!m_mem.addr_exists(addr)) begin
                         |
xmvlog: *E,NOTCLM (/home/rmasumda/projects/ibex_riscv/dv/uvm/core_ibex/common/ibex_mem_intf_agent/ibex_mem_intf_response_seq_lib.sv,180|25): addr_exists is not a class item.
    if (!m_mem.addr_exists(addr)) begin
                         |
xmvlog: *E,NOTFXX (/home/rmasumda/projects/ibex_riscv/dv/uvm/core_ibex/common/ibex_mem_intf_agent/ibex_mem_intf_response_seq_lib.sv,180|25): Expecting a function name [10.3.3(IEEE)]

These are the first 2 error messages followed by 

  `DV_CREATE_SIGNAL_PROBE_FUNCTION(signal_probe_rf_ren_a, rf_ren_a)
                                 |
xmvlog: *E,NOTDIR (/home/rmasumda/projects/ibex_riscv/dv/uvm/core_ibex/env/core_ibex_dut_probe_if.sv,92|33): `DV_CREATE_SIGNAL_PROBE_FUNCTION: not a recognized directive or macro [2.7.3][16.3.1][16(IEEE)].

rswarbrick, 2023-04-20:
Ah, well the missing macro is also caused by an ordering problem. I think VCS is internally concatenating files in a different order from what Xcelium happens to choose. That this causes a problem is a bug in our code! :-D Fortunately, the fix is easy. If you look at e.g. `clk_rst_if.sv`, you'll see it include the definition we need. I think you can just copy/paste some of that code and add something like this to `core_ibex_dut_probe_if.sv`:
```
`ifndef VERILATOR
  `include "dv_macros.svh"
`endif
```
If that fixes things, we should take both changes that we've found so far!

For the other error, I'm a bit confused. The message looks like it's saying that it doesn't know that `addr_exists` makes sense here. But I think that `m_mem` is of type `mem_model` and it looks like `mem_model.svh` defines that function (on line 96). The message about `DV_CREATE_SIGNAL_PROBE_FUNCTION` is also rather confusing. The macro is defined in `dv_macros.svh`. Maybe something had already gone awry before the error message came out?

So I'm sorry that I haven't properly fixed things for you with Xcelium. If you're still finding issues there, is there any chance you could give a step-by-step sequence for us to reproduce the error? Thanks!

raiyyanfaisal09, 2023-04-28:
Hi @rswarbrick, Thanks for the response. That definitely helped me get over the first issue I was seeing. However, I wasn't able to solve the issue with mem_model call. I also forgot to mention, I was seeing an error related to the DPI calls in ibex_if_stage.sv file (that I suspect is a DPI call related to icache?) before I saw this error I mentioned above & since we are not using icache at this moment I thought of commenting it out. Thats when I saw the issue I mentioned above related to m_mem_model. 

So that brings me to my question - what could possibly be causing an issue there ? 

Also, I pulled the latest yesterday and I don't remember the commit number in which I was seeing those errors. Hence, I am seeing totally new set of errors now. 

This is what I am currently seeing with VCS as the simulator : 

Error-[FCIWCS] Invalid 'with' clause specification
/home/rmasumda/projects/ibex_riscv/dv/uvm/core_ibex/fcov/core_ibex_fcov_if.sv, 646
core_ibex_fcov_if, "priv_mode_irq_cross"
  The 'with' clause specified on the cross bin does not have any impact and is
  invalid; it should contain at least one occurence of the constituent 
  coverpoint names.
  Please remove or correct the clause and compile

Syntactically it looks fine, however I don't see definition of some of the enum data type variables. I am not sure if that is really the issue, I am only suspecting here. The reason being, greping the variable 'PRIV_LVL_M' in 'dv' only shows me the instances where it is used and not the actual definition (if at all the error is actually related to this variable and not something else). 

Would really appreciate if you can give me some insight into how to solve this issue. 

I have also been wondering, if I am supposed to be pulling a certain commit that is clean and tested and not do so randomly. If you do recommend to take in the new updates that is clean and tested then is there a way to know what and when to pull in those updates? 

Thanks!

rswarbrick, 2023-07-12:
Hi there! It sounds like things aren't properly installed. In particular, `riscv/devices.h`  should appear in the `include` directory of a Spike installation. I'd suggest looking through your installation and working out where that file is. (I did the stupid option here of `find /opt/spike -name devices.h`, but obviously you'll need a different root path)