lowRISC/ibex #2012: Ibex simple system

PZZONE, 2023-04-09:
***Observed Behavior***
————————————————
The xrun command fails with:
```shell
$ make build-simple-system
fusesoc --cores-root=. run --target=sim --setup --build \
        lowrisc:ibex:ibex_simple_system \
        --RV32E=0 --RV32M=ibex_pkg::RV32MFast --RV32B=ibex_pkg::RV32BNone --RegFile=ibex_pkg::RegFileFF --BranchTargetALU=0 --WritebackStage=0 --ICache=0 --ICacheECC=0 --ICacheScramble=0 --BranchPredictor=0 --DbgTriggerEn=0 --SecureIbex=0 --PMPEnable=0 --PMPGranularity=0 --PMPNumRegions=4 --MHPMCounterNum=0 --MHPMCounterWidth=40
INFO: Preparing lowrisc:dv:crypto_prince_ref:0.1
INFO: Preparing lowrisc:dv:dv_fcov_macros:0
INFO: Preparing lowrisc:dv:secded_enc:0
INFO: Preparing lowrisc:dv_verilator:ibex_pcounts:0
INFO: Preparing lowrisc:dv_verilator:simutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_pkg:0.1
INFO: Preparing lowrisc:prim:primgen:0.1
INFO: Preparing lowrisc:prim:ram_1p_pkg:0
INFO: Preparing lowrisc:prim:ram_2p_pkg:0
INFO: Preparing lowrisc:prim:util_get_scramble_params:0
INFO: Preparing lowrisc:tool:check_tool_requirements:0.1
INFO: Preparing lowrisc:dv:scramble_model:0
INFO: Preparing lowrisc:dv_verilator:memutil_dpi:0
INFO: Preparing lowrisc:lint:common:0.1
INFO: Preparing lowrisc:prim:prim_pkg:0.1
INFO: Preparing lowrisc:dv_verilator:memutil_dpi_scrambled:0
INFO: Preparing lowrisc:dv_verilator:memutil_verilator:0
INFO: Preparing lowrisc:prim:assert:0.1
INFO: Preparing lowrisc:prim:buf:0
INFO: Preparing lowrisc:prim:cipher_pkg:0.1
INFO: Preparing lowrisc:prim:clock_gating:0
INFO: Preparing lowrisc:prim:clock_mux2:0
INFO: Preparing lowrisc:prim:flop:0
INFO: Preparing lowrisc:prim:ram_1p:0
INFO: Preparing lowrisc:prim:ram_2p:0
INFO: Preparing lowrisc:prim:secded:0.1
INFO: Preparing lowrisc:ibex:ibex_icache:0.1
INFO: Preparing lowrisc:ibex:ibex_tracer:0.1
INFO: Preparing lowrisc:ibex:sim_shared:0
INFO: Preparing lowrisc:prim:cipher:0
INFO: Preparing lowrisc:prim:lfsr:0.1
INFO: Preparing lowrisc:prim:util:0.1
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:prim:onehot_check:0
INFO: Preparing lowrisc:prim:ram_1p_adv:0.1
INFO: Preparing lowrisc:prim:ram_1p_scr:0.1
INFO: Preparing lowrisc:ibex:ibex_top:0.1
INFO: Preparing lowrisc:ibex:ibex_top_tracing:0.1
INFO: Preparing lowrisc:ibex:ibex_simple_system_core:0
INFO: Preparing lowrisc:ibex:ibex_simple_system:0
INFO: Generating lowrisc:prim:prim_pkg-impl:0.1
Creating prim_pkg.sv
Core file written to prim_pkg.core.
INFO: Generating lowrisc:prim:buf-impl:0
Implementations for primitive buf: generic, xilinx
Inspecting generic module /home/ysyx/project/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_buf.sv
**No module named 'anytree'**
**Verible parser failed, using regex fallback instead.**
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_buf-impl_0/prim_buf.sv
Creating core file for primitive buf.
Core file written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_buf-impl_0/prim_buf.core
INFO: Generating lowrisc:prim:clock_gating-impl:0
Implementations for primitive clock_gating: generic, xilinx
Inspecting generic module /home/ysyx/project/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_clock_gating.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.sv
Creating core file for primitive clock_gating.
Core file written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.core
INFO: Generating lowrisc:prim:clock_mux2-impl:0
Implementations for primitive clock_mux2: generic, xilinx
Inspecting generic module /home/ysyx/project/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_clock_mux2.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_mux2-impl_0/prim_clock_mux2.sv
Creating core file for primitive clock_mux2.
Core file written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_mux2-impl_0/prim_clock_mux2.core
INFO: Generating lowrisc:prim:flop-impl:0
Implementations for primitive flop: generic, xilinx
Inspecting generic module /home/ysyx/project/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_flop.sv
**No module named 'anytree'**
**Verible parser failed, using regex fallback instead.**
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_flop-impl_0/prim_flop.sv
Creating core file for primitive flop.
Core file written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_flop-impl_0/prim_flop.core
INFO: Generating lowrisc:prim:ram_1p-impl:0
Implementations for primitive ram_1p: generic, badbit
Inspecting generic module /home/ysyx/project/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_ram_1p.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_1p-impl_0/prim_ram_1p.sv
Creating core file for primitive ram_1p.
Core file written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_1p-impl_0/prim_ram_1p.core
INFO: Generating lowrisc:prim:ram_2p-impl:0
Implementations for primitive ram_2p: generic
Inspecting generic module /home/ysyx/project/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_ram_2p.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_2p-impl_0/prim_ram_2p.sv
Creating core file for primitive ram_2p.
Core file written to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_2p-impl_0/prim_ram_2p.core
INFO: Wrote dependency graph to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/lowrisc_ibex_ibex_simple_system_0.deps-after-generators.dot
INFO: Wrote Makefile fragment to /home/ysyx/project/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/core-deps.mk
INFO: Setting up project

INFO: Running pre_build script check_tool_requirements
INFO: Building simulation model
``` 
this problem is :
**"No module named 'anytree"
"Verible parser failed, using regex fallback instead."**
and then i running this command
``` shell 
$ make -C examples/sw/simple_system/hello_test
$ ./build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/Vibex_simple_system [-t] --meminit=ram,./examples/sw/simple_system/hello_test/hello_test.elf
Simulation of Ibex
==================

Tracing can be toggled by sending SIGUSR1 to this process:
$ kill -USR1 859329

Simulation running, end by pressing CTRL-c.
TOP.ibex_simple_system.u_top.u_ibex_tracer.printbuffer_dumpline.unnamedblk1: Writing execution trace to trace_core_00000000.log
Terminating simulation by software request.
- ../src/lowrisc_ibex_sim_shared_0/./rtl/sim/simulator_ctrl.sv:93: Verilog $finish
Received $finish() from Verilog, shutting down simulation.

Simulation statistics
=====================
Executed cycles:  13138
Wallclock time:   0.074 s
Simulation speed: 177541 cycles/s (177.541 kHz)

Performance Counters
====================
Cycles:                     475
NONE:                       0
Instructions Retired:       261
LSU Busy:                   0
Fetch Wait:                 0
Loads:                      0
Stores:                     0
Jumps:                      0
Conditional Branches:       0
Taken Conditional Branches: 0
Compressed Instructions:    0
Multiply Wait:              0
Divide Wait:                0
``` 

****My Environment****
**Operating system:** Ubuntu 22.04
**Version of the Ibex source code:**
latest master branch 93c8e92
**fusesoc** /home/ysyx/.local/bin fusesoc 0.1

****Steps to reproduce the issuet****
I have build the env for this (https://github.com/lowRISC/ibex/tree/master/examples/simple_system)

PZZONE, 2023-04-11:
> The first message (about "anytree") is to do with how our tooling spots error messages. The message basically says "I don't think you have an add-on to Verible that you'd need, so I'm doing something simpler to parse the output".
> 
> Is there anything else that looks like an error to you? This looks to me like everything worked!
Thank you very much!!!

PZZONE, 2023-04-11:
> The first message (about "anytree") is to do with how our tooling spots error messages. The message basically says "I don't think you have an add-on to Verible that you'd need, so I'm doing [...]

***Thank you so much for answering my question! ! !*** 
I have installed Verible binary files before, and directly imported tools such as verible-* into the /opt/riscv/ path by decompression.
The result of executing which verible-verilog-lint &  verible-verilog-lint --version:
[image]
All verible tools are as follows:
[image]
So is this problem caused by incomplete configuration of my Verible environment?

PZZONE, 2023-04-11:
I have to install the Verible by bazel for (https://github.com/chipsalliance/verible)
[image]
and then i executing the command:
```
verible-verilog-lint --version
v0.0-3132-gb052efb1
Commit  2023-04-03
Built   2023-04-11T10:31:04Z
```
[image]

but when i run the command (make build-simple-system) , i have the same information :
"No module named 'anytree"
"Verible parser failed, using regex fallback instead."

When I searched for related issues, I found that there had been similar issues in the opentitan project. (This problem is caused by the version of fusesoc) I don't know how I need to find this problem, or try to modify it, can you give me some hints?
（https://github.com/lowRISC/opentitan/issues/12769）

@rswarbrick ,Sorry to bother you again

rswarbrick, 2023-04-13:
Hi again! To be clear, this message isn't reporting a real problem: it's just saying that the tooling thinks it might flag some spurious errors (but it seems that it doesn't!). If this is the only thing you're noticing, there's not really anything for you to do. If there's a separate problem that you're seeing, maybe it's worth opening a new issue to avoid the problem getting hidden here.

closed 2023-04-13 by rswarbrick

PZZONE, 2023-04-13:
> Hi again! To be clear, this message isn't reporting a real problem: it's just saying that the tooling thinks it might flag some spurious errors (but it seems that it doesn't!). If this is the only [...]

Thank you very much ！！！
I guess my problem is that I didn't set reasonable parameters;
reference to this https://github.com/lowRISC/ibex/issues/1866;