lowRISC/ibex #2018: Ibex and latest riscv toolchains: error "unrecognized opcode csrr"

RTLhamster, 2023-04-20:
Hi,

I run into a problem with my riscv toolchain and ibex.
I'm using the latest pre-compiled riscv32-gcc-ubuntu2004-release/4/ toolchain from embecosm and I get a assembly error:
`interrupt.h: Assembler messages:
`
`interrupt.h:83: Error: unrecognized opcode csrr t3,769, extension zicsr required
`

The interrupt.h file is using csrr / csrw commands for setting up interrupts of the ibex.
I'm using the arch.-flag `-march=rv32imc` but also tried `-march=rv32imczicsr` variants, it doesn't change.

The strange thing is, that my software setup was working for older toolchains (installed in october 22), but when I switch to the latest ones (the COREV version doesn't work either), it fails.

Did anyone face a similar problem?
Maybe more general: which toolchain do you IBEX people suggest to use?
(I also compiled the https://github.com/riscv-collab/riscv-gnu-toolchain my self, but it takes ages, 
therefore switched to embecosm toolchains)

Thank you!

## My Environment
Ibex core in maxperf configuration
but its more a toolchain problem

**Operating system:**
Ubuntu 22.04.

GregAC, 2023-04-21:
Have you tried the lowRISC toolchain builds: https://github.com/lowRISC/lowrisc-toolchains/releases/tag/20220524-1?

RTLhamster, 2023-04-21:
Hi GregAC,
not yet. I was reluctant to try the lowRISC one since its stating 
"The toolchains are not supported by lowRISC or recommended to be used outside of lowRISC."
I'm actually also using the cv32e40p, therefore I want to go with a more standard compiler.

closed 2023-05-19 by rswarbrick