lowRISC/ibex #2022: fence instruction throwing unexpected illagal in RV32E congfiguration

tzwaenn, 2023-04-21:
## Observed Behavior
In a configuration with RV32E, I see an illegal exception being thrown for a fence instruction. While this instruction has fields rs1 and rd, the spec clarifies
"The unused fields in the FENCE instructions—rs1 and rd—are reserved for finer-grain fences in future extensions. For forward compatibility, base implementations shall ignore these fields, and standard software shall zero these fields."
So it is clearly not intended that a fence with rs1[4]==1 or rd[4]==1 should throw an illegal exception on RV32E - it is not accessing any integer registers.

**EDA tool and version:**
The issue was found with the Siemens Processor Verification App.

**Version of the Ibex source code:**
033abfc09faa9c208af257acb30fa15aa011f3c3 with RTL parameter RV32E=1

GregAC, 2023-04-25:
Thanks for the report @tzwaenn. If you have a fix feel free to submit a PR otherwise we will take a look though this isn't a priority for us right now (current development focus is on the opentitan configuration which has E=0).

tzwaenn, 2023-04-25:
Understood. I think removing line 1145 in ibex_decoder.sv
alu_op_a_mux_sel_o = OP_A_REG_A;
would do the trick (fence does not use the source register), though I also do not  have the bandwidth right now to run experiments.