lowRISC/ibex #2026: Ibex Simple System Riscv-tests support

kevinhe5, 2023-05-08:
Hi, I'm new to Ibex and have been trying to use simple system with Verilator (version 3.922). I tried the run the simulator binary with [riscv-tests](https://github.com/riscv-software-src/riscv-tests/tree/79064081503b53fdb44094e32ff54a3ab20a9bf2) but when I run the simplest rv32ui elf with simple system, I get an output with all illegal instructions. Does Ibex Simple System support riscv-tests out of the box or will I have to make some modification? Just wondering if anyone had setup Ibex for this before. Thanks!
## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

rswarbrick, 2023-05-11:
Hi there! The problem that you're running into is the reset vector address. I've just tried to build the same thing and I think that the `riscv-tests` repository assumes that the reset vector is at `0x80000000`. Ibex actually starts at `0x00100000`, which explains why it's getting random rubbish instead of instructions!

You will either need to generate an ELF with the memory layout that Spike is expecting (by hacking on the `riscv-tests` code) or you'll need to tell Spike's `simple_system` to use a different memory layout. That's probably harder, though: I'd suggest fixing the test ELF file.

closed 2023-05-11 by rswarbrick

kevinhe5, 2023-05-15:
Is there anyway to set the reset vector to `0x80000000` through a parameter or config?