lowRISC/ibex #2027: How can I generate two different .vmem files for two different memory (I & D)

abolfazlsajadi, 2023-05-09:
## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

rswarbrick, 2023-05-11:
Hi there! I think we'll need a little more information to be able to help you. If I understand correctly, you're building an image and loading it into a simulated Ibex, but want to generate two different vmem files to use for initialising the I and D sides. Is that right?

If I'm not quite right, could you give slightly more detail about what you're doing, and what you would expect to happen?

Many thanks!

abolfazlsajadi, 2023-05-12:
Greetings and thank you for your response
Suppose I have two separate memory cells for instructions and data, and I have two ways of separating them from each other: 1) create two VMEM files that begin at address zero, and 2) control data between the two according to one unusable bit. Based on my 64 kbyte memory, I need a 16-bit address, so that I can assign the bit of 16 to control data between two memories, in other words, if data begins at 0x10000, it will be placed in data memory, and the bit of 16 will be removed from the address bus. in order to do that i change the linker file like this:
`/* Copyright lowRISC contributors.
   Licensed under the Apache License, Version 2.0, see LICENSE for details.
   SPDX-License-Identifier: Apache-2.0 */

OUTPUT_ARCH(riscv)

MEMORY
{
    /* 60 kB should be enough for anybody... */
    ram         : ORIGIN = 0x00100000, LENGTH = 0xFFFF  
    Datamem       : ORIGIN = 0x00010000, LENGTH = 0x1FFFF
    stack       : ORIGIN = 0x00020000, LENGTH = 0x2000   
}

/* Stack information variables */
_min_stack      = 0x1000;   /* 4K - minimum stack space to reserve */
_stack_len     = LENGTH(stack);
_stack_start   = ORIGIN(stack) + LENGTH(stack);

_entry_point = _vectors_start + 0x80;
ENTRY(_entry_point)

SECTIONS
{
    .text : {
        . = ALIGN(4);
		_vectors_start = .;
        KEEP(*(.vectors))
		_vectors_end = .;
        *(.text)
        *(.text.*)
        . = ALIGN(4);
    }  > ram

    .rodata : {
        . = ALIGN(4);
        /* Small RO data before large RO data */
        *(.srodata)
        *(.srodata.*)
        *(.rodata);
        *(.rodata.*)
        . = ALIGN(4);
    } > Datamem

    .data : {
        . = ALIGN(4);
        /* Small data before large data */
        *(.sdata)
        *(.sdata.*)
        *(.data);
        *(.data.*)
        . = ALIGN(4);
    } > Datamem

    .bss :
    {
        . = ALIGN(4);
        _bss_start = .;
        /* Small BSS before large BSS */
        *(.sbss)
        *(.sbss.*)
        *(.bss)
        *(.bss.*)
        *(COMMON)
        _bss_end = .;
        . = ALIGN(4);
    } > Datamem

    /* ensure there is enough room for stack */
    .stack (NOLOAD): {
        . = ALIGN(4);
        . = . + _min_stack ;
        . = ALIGN(4);
        stack = . ;
        _stack = . ;
    } > stack
}`
Although it is correct in the map and ELF files, the addresses have changed after conversion to binary files, and I am not aware of what has happened.

rswarbrick, 2023-05-19:
Ahah, now I understand, thanks. The linker script looks sensible to me, I think. So you're saying that the addresses are correct in the ELF files, but become wrong when the ELF is converted to a binary?

That seems confusing to me! After all, the "conversion" process just consists of throwing away headers. If you're sure that's what's happening, maybe it would make sense to attach an example ELF and binary to the report? We could take a look and try to work out what's going wrong.

(Note: This isn't really an Ibex bug at this point - it seems to be a problem with using the toolchain)

abolfazlsajadi, 2023-06-22:
Hello, it was my fault.

The problem has been resolved

thanks

closed 2023-06-22 by abolfazlsajadi