lowRISC/ibex #2031: Non-reset registers in Latch-based RF cause X values propagating during LW instructions.

BlooHoosker, 2023-05-11:
## Observed Behavior

Code in question:
```
  // Actual write operation:
  // Generate the sequential process for the NUM_WORDS words of the memory.
  // The process is synchronized with the clocks mem_clocks[i], i = 1, ..., NUM_WORDS-1.
  for (genvar i = 1; i < NUM_WORDS; i++) begin : g_rf_latches
    always_latch begin
      if (mem_clocks[i]) begin
        mem[i] = wdata_a_q;
      end
    end
  end
```

When running IBEX with LATCH register file at the start of the simulation there are X values in those LATCH registers because they are not reset by rst_n.

When LW instruction is being executed at the start, when register file is still filled with X values, it causes X values to propagate all the way to data_wdata_o at the TOP.  

While the LW instruction does not use the second register operand (and there is just imm), part of the imm value still propagates all the way to the register file into the following assignment:
assign rdata_b_o = mem[raddr_b_int];

This (at the start) selects a register from the RF which contains X values and they propagate further.

As a side effect this also triggers the IbexDataPayloadX

## Expected Behavior
Either Latches to be reset not contain X values at the start or not propagate the X values outside and trigger the assert.

Or should I setup the simulator so that the latches have some value at the start despite not having reset?

## Steps to reproduce the issue
Execute LW instruction before registers in latch RF are written into and contain some value.

## My Environment
RV32E configuration of IBEX with Latch register file.

**EDA tool and version:**
XCELIUM 21.09.007

**Version of the Ibex source code:**
Latest

rswarbrick, 2023-05-19:
Hi there! This sounds like the reset vector needs a tweak: if we're not coming out of reset with known values in the registers, they probably need to be zeroed explicitly at the start. What firmware are you running?

For an example, run `make` in `examples/sw/simple_system/hello_test` and look at the generated binary. If you use objdump, you'll see that the reset vector looks like this:
```
00100346 <reset_handler>:
  100346:       00000093                li      ra,0
  10034a:       8106                    mv      sp,ra
  10034c:       8186                    mv      gp,ra
  10034e:       8206                    mv      tp,ra
  100350:       8286                    mv      t0,ra
  100352:       8306                    mv      t1,ra
  100354:       8386                    mv      t2,ra
  100356:       8406                    mv      s0,ra
  100358:       8486                    mv      s1,ra
  10035a:       8506                    mv      a0,ra
  10035c:       8586                    mv      a1,ra
  10035e:       8606                    mv      a2,ra
  100360:       8686                    mv      a3,ra
  100362:       8706                    mv      a4,ra
  100364:       8786                    mv      a5,ra
  100366:       8806                    mv      a6,ra
  100368:       8886                    mv      a7,ra
  10036a:       8906                    mv      s2,ra
  10036c:       8986                    mv      s3,ra
  10036e:       8a06                    mv      s4,ra
  100370:       8a86                    mv      s5,ra
  100372:       8b06                    mv      s6,ra
  100374:       8b86                    mv      s7,ra
  100376:       8c06                    mv      s8,ra
  100378:       8c86                    mv      s9,ra
  10037a:       8d06                    mv      s10,ra
  10037c:       8d86                    mv      s11,ra
  10037e:       8e06                    mv      t3,ra
  100380:       8e86                    mv      t4,ra
  100382:       8f06                    mv      t5,ra
  100384:       8f86                    mv      t6,ra
  100386:       00038117                auipc   sp,0x38
  10038a:       c7a10113                addi    sp,sp,-902 # 138000 <_stack_start>
```
We didn't do this by hand: it's built into the RISC-V toolchain that we're using.

Maybe you could reply with some more details of exactly what you're doing?

BlooHoosker, 2023-05-19:
Hello, thank you for your reply.

I see, so with the LATCH register file it is expected to initialize the values after reset by software? I do not use such reset vector software like the one which you included in your comment.

The system I am building used another RISC-V CPU which has its RF reset to zeros with system reset signal.

Now I understand that I need to have a reset vector SW for the latch register file. 

Thanks!

rswarbrick, 2023-05-22:
Ah, ok, that makes sense. Thanks for the reply.

Ibex isn't that unusual here: I think that lots of processors expect some of the "initialisation of state" to be done by firmware. With processors I've worked on in the past, we had lots of "bootstrapping" in assembly to do to get into a known state at the start of time. It seemed a bit ridiculous to be debugging a boot vector that I wrote, but I've had to do that at least once!

closed 2023-05-22 by rswarbrick