lowRISC/ibex #2039: Report a bug in Ibex

AnantaSrikar, 2023-05-22:
## Observed Behavior
Simulator fails to build with the following output:
```
test@riscv-dev:~/ibex$ fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast
INFO: Preparing lowrisc:dv:crypto_prince_ref:0.1
INFO: Preparing lowrisc:dv:dv_fcov_macros:0
INFO: Preparing lowrisc:dv:secded_enc:0
INFO: Preparing lowrisc:dv_verilator:ibex_pcounts:0
INFO: Preparing lowrisc:dv_verilator:simutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_pkg:0.1
INFO: Preparing lowrisc:prim:primgen:0.1
INFO: Preparing lowrisc:prim:ram_1p_pkg:0
INFO: Preparing lowrisc:prim:ram_2p_pkg:0
INFO: Preparing lowrisc:prim:util_get_scramble_params:0
INFO: Preparing lowrisc:tool:check_tool_requirements:0.1
INFO: Preparing lowrisc:dv:scramble_model:0
INFO: Preparing lowrisc:dv_verilator:memutil_dpi:0
INFO: Preparing lowrisc:lint:common:0.1
INFO: Preparing lowrisc:prim:prim_pkg:0.1
INFO: Preparing lowrisc:dv_verilator:memutil_dpi_scrambled:0
INFO: Preparing lowrisc:dv_verilator:memutil_verilator:0
INFO: Preparing lowrisc:prim:assert:0.1
INFO: Preparing lowrisc:prim:buf:0
INFO: Preparing lowrisc:prim:cipher_pkg:0.1
INFO: Preparing lowrisc:prim:clock_gating:0
INFO: Preparing lowrisc:prim:clock_mux2:0
INFO: Preparing lowrisc:prim:flop:0
INFO: Preparing lowrisc:prim:ram_1p:0
INFO: Preparing lowrisc:prim:ram_2p:0
INFO: Preparing lowrisc:prim:secded:0.1
INFO: Preparing lowrisc:ibex:ibex_icache:0.1
INFO: Preparing lowrisc:ibex:ibex_tracer:0.1
INFO: Preparing lowrisc:ibex:sim_shared:0
INFO: Preparing lowrisc:prim:cipher:0
INFO: Preparing lowrisc:prim:lfsr:0.1
INFO: Preparing lowrisc:prim:util:0.1
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:prim:onehot_check:0
INFO: Preparing lowrisc:prim:ram_1p_adv:0.1
INFO: Preparing lowrisc:prim:ram_1p_scr:0.1
INFO: Preparing lowrisc:ibex:ibex_top:0.1
INFO: Preparing lowrisc:ibex:ibex_top_tracing:0.1
INFO: Preparing lowrisc:ibex:ibex_simple_system_core:0
INFO: Preparing lowrisc:ibex:ibex_simple_system:0
INFO: Generating lowrisc:prim:prim_pkg-impl:0.1
Creating prim_pkg.sv
Core file written to prim_pkg.core.
INFO: Generating lowrisc:prim:buf-impl:0
Implementations for primitive buf: xilinx, generic
Inspecting generic module /home/test/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_buf.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/test/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_buf-impl_0/prim_buf.sv
Creating core file for primitive buf.
Core file written to /home/test/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_buf-impl_0/prim_buf.core
INFO: Generating lowrisc:prim:clock_gating-impl:0
Implementations for primitive clock_gating: generic, xilinx
Inspecting generic module /home/test/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_clock_gating.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/test/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.sv
Creating core file for primitive clock_gating.
Core file written to /home/test/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.core
INFO: Generating lowrisc:prim:clock_mux2-impl:0
Implementations for primitive clock_mux2: generic, xilinx
Inspecting generic module /home/test/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_clock_mux2.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
[... 59 lines cut ...]
                                                                              : ... Suggest using delayed assignment '<='
  119 |       file_handle = $fopen(file_name, "w");
      |                   ^
%Error: Exiting due to 3 warning(s)
make: *** [Makefile:16: Vibex_simple_system.mk] Error 1

ERROR: Failed to build lowrisc:ibex:ibex_simple_system:0 : '['make', '-j', '8']' exited with an error: 2
```

## Expected Behavior
The simulator builds successfully.

## Steps to reproduce the issue:
1. Install all requirements.
2. Run the build command for [SimpleSystem](https://github.com/lowRISC/ibex/tree/master/examples/simple_system): `fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast`
3. You'll then run into the error.

## My Environment
Installed all the requirements as per specified [prerequisites](https://github.com/lowRISC/ibex/tree/master/examples/simple_system#prerequisites)

**EDA tool and version: NA**

**Operating system: Ubuntu 22.04.2 LTS**

**Version of the Ibex source code: a31c043**

rswarbrick, 2023-05-22:
Thanks for the report. As we said on Zulip, I suspect the problem is probably your Verilator version. I'm going to close this issue now, because I don't *think* there's anything more that we can do to help. Please re-open it if I'm wrong! :-)

closed 2023-05-22 by rswarbrick

yulong-lan, 2023-05-27:
hallo Ananta, have you figured out then problem you encountered? I have the latest version of verilator and I run into the same problem as you did.

AnantaSrikar, 2023-05-27:
> hallo Ananta, have you figured out then problem you encountered? I have the latest version of verilator and I run into the same problem as you did.

Yes, turns out the latest development version of verilator isn't compatible with ibex. I used Verilator version `4.220` and then everything worked out as expected. Hope that helped!

yulong-lan, 2023-05-27:
Thank you for your quick reply, i am compiling this version now and will install it and try. By the way, I am trying to write a master thesis involving the use of ibex and other open-source processors, but I am a beginner and hope to get some help. Is there any way I can send you messages and chat, if you are ok with it?

AnantaSrikar, 2023-05-27:
> Thank you for your quick reply, i am compiling this version now and will install it and try. By the way, I am trying to write a master thesis involving the use of ibex and other open-source [...]

Yup works. HMU on my discord: `epoch101#8909`. However I'm pretty new to ibex too, but would be glad if I could be of any help!

BMorgan1296, 2023-07-27:
I just had this same issue, it is working with Verilator v4.228. Must be a v5.x issue.