lowRISC/ibex #2049: make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 255

athurwo, 2023-06-16:
Hi,
checkout ibex code, just change simulator to vcs in Makefile.

when make in  ibex/dv/uvm/core_ibex,   then occur a error like below:
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 255
make: *** [Makefile:66: run] Error 2

the code of scripts/ibex_sim.mk:42 is : $(verb)env PYTHONPATH=$(PYTHONPATH) \
$(verb) is @, I don`t know whether "env" is correct or not, but I can`t find "env" in GNU make document.

when I change env to set in all makefile,  then can run flow like below:
Running RTL simulation at /home/fso/ibex/dv/uvm/core_ibex/out/run/tests/riscv_bitmanip_balanced_test.9179
Collecting simulation results and checking logs of testcase at /home/fso/ibex/dv/uvm/core_ibex/out/run/tests/riscv_bitmanip_balanced_test.9179/trr.yaml
Collecting up results of tests into report regr.log

but I can`t find regr.log ,  the conext of trr.yaml is below:

passed:                 
failure_mode:           
failure_message:        
timeout_s:              
testtype:               TestType.RISCVDV
testdotseed:            riscv_bitmanip_balanced_test.6718
testname:               riscv_bitmanip_balanced_test
seed:                   6718
binary:                 
rtl_simulator:          vcs
iss_cosim:              spike
gen_test:               
gen_opts:               
rtl_test:               
sim_opts:               
directed_data:          
dir_test:               .
assembly:               
objectfile:             
riscvdv_run_gen_log:    
riscvdv_run_gen_stdout: 
riscvdv_run_log:        
riscvdv_run_stdout:     
compile_asm_gen_log:    
compile_asm_log:        
rtl_log:                
rtl_stdout:             
rtl_trace:              
iss_cosim_log:          
iss_cosim_trace:        
dir_fcov:               
riscvdv_run_gen_cmds:   
riscvdv_run_cmds:       
compile_asm_gen_cmds:   
compile_asm_cmds:       
rtl_cmds:               
metadata_pickle_file:   /home/fso/ibex/dv/uvm/core_ibex/out/metadata/metadata.pickle
pickle_file:            /home/fso/ibex/dv/uvm/core_ibex/out/metadata/riscv_bitmanip_balanced_test.6718.pickle
yaml_file:              trr.yaml

the passed is blank.

Please help me resovle it.
thank you.

athurwo, 2023-06-19:
> Hmm, I'm not sure exactly what's going on here, but a bit of background might help: `env` is a standard linux command (man page [here](https://www.man7.org/linux/man-pages/man1/env.1.html)) which sets an environment variable for the binary that's being run. We missed some indentation just below (oops!), but that command is just running `compile_tb.py` with PYTHONPATH set to something sensible. I doubt the environment variable is connected to the failure you're seeing.
> 
> I'm not sure how to debug further without you trying some more stuff, but I think you should be able to find some generated files that show what is going on in `/home/fso/ibex/dv/uvm/core_ibex/out`.

I have resovled it. this is toolchain not have riscv b extension and some issue with vcs2016, not makefile issue, I modify sytanx code to meet vcs2016, but now some problem need high version of vcs, so I need to install vcs2020-03-sp2.

rswarbrick, 2023-06-19:
Cool, thanks.

closed 2023-06-19 by rswarbrick