lowRISC/ibex #2054: Unexpected PoR value of mstatus

MikeOpenHWGroup, 2023-06-22:
### Bug Description

The [documented](https://ibex-core.readthedocs.io/en/latest/03_reference/cs_registers.html#machine-status-mstatus) power-on-reset value of the `mstatus` CSR (0x300) is 0x0000_1800.  The actual PoR value is 0x0000_0080.

I believe that the documented PoR value is the proper one, but the coded PoR has been in the Ibex code for three years so this is as much a question as it is a bug report.

## Observed Behavior

In simulations of our fork of Ibex, coming out of reset, the read value of `mstatus` is 0x0000_0080.  This is matches the coded PoR in the [RTL](https://github.com/lowRISC/ibex/blob/a7845832a2895ac3be8e28f41b87992207daee30/rtl/ibex_cs_registers.sv#L859):
```
localparam status_t MSTATUS_RST_VAL = '{mie:  1'b0,
                                          mpie: 1'b1,
                                          mpp:  PRIV_LVL_U,
                                          mprv: 1'b0,
                                          tw:   1'b0};
```
Given that [PRIV_LVL_U](https://github.com/lowRISC/ibex/blob/a7845832a2895ac3be8e28f41b87992207daee30/rtl/ibex_pkg.sv#L208) is 2'b00, this yields a PoR of 0x0000_0080.

## Expected Behavior

Coming out of reset, the read value of `mstatus` should be 0x0000_1800.  The RTL definition should look like this:
```
localparam status_t MSTATUS_RST_VAL = '{mie:  1'b0,
                                          mpie: 1'b0,
                                          mpp:  PRIV_LVL_M,
                                          mprv: 1'b0,
                                          tw:   1'b0};
```

## Steps to reproduce the issue

I have a test-program in CORE-V-VERIF for the CVE2, but do not have an Ibex test for this.  Having said that, visual inspection easily supports what we are seeing in simulation.

## My Environment

The OpenHW Group has created a fork of the Ibex which we are branding as the [CVE2](https://github.com/openhwgroup/cve2/tree/main).  The first release of this project will be the CV32E20.

**EDA tool and version:**

Questasim 2021.4_4

**Operating system:**

CentOS 7.9.2009

**Version of the Ibex source code:**

The current head as of 2023-06-22 @ 10:35 EDT: `a784583`.

GregAC, 2023-06-29:
Thanks for flagging this @MikeOpenHWGroup I think we can call this a documentation issue.

I'd note the RISC-V privileged spec has this to say (under 3.4 Reset):

> Upon reset, a hart’s privilege mode is set to M. The mstatus fields MIE and MPRV are reset to 0.  If little-endian memory accesses are supported, the mstatus/mstatush field MBE is reset to 0. The misa register is reset to enable the maximal set of supported extensions and widest MXLEN, as described in Section 3.1.1. For implementations with the “A” standard extension, there is no valid load reservation. The pc is set to an implementation-defined reset vector. The mcause register is set to a value indicating the cause of the reset. Writable PMP registers’ A and L fields are set to 0, unless the platform mandates a different reset value for some PMP registers’ A and L fields. If the hypervisor extension is implemented, the hgatp.MODE and vsatp.MODE fields are reset to 0.  No WARL field contains an illegal value. All other hart state is unspecified.

The final 'All other hart state is unspecified' sentence being the key one.

So in this instance either `PRIV_LVL_M` or `PRIV_LVL_U` are acceptable PoR values. So better to keep the RTL as is and correct the documentation I think.

MikeOpenHWGroup, 2023-06-29:
That works for me @GregAC, thanks for looking into this.

Feel free to close this issue whenever the documentation is updated.  If your protocol on Issue management dictates that only the Originator (me, in this case) can close an issue, just @ mention me and I'll be glad to.

GregAC, 2023-07-03:
I've put a fix together #2056 thanks @MikeOpenHWGroup

commit 96988c6b (2023-07-04): [doc] Fix documented mstatus reset value

Fixes #2054
PR #2056 (2023-07-04): [doc] Fix documented mstatus reset value
Fixes #2054
closed 2023-07-04 by GregAC