lowRISC/ibex #2055: dv: building testbench fails

RanjanThales, 2023-06-30:
Hi team,
Previously, I was running simulation of ibex smoothly, however now i am facing some issue related to building testbench  (with new makefile)

## My Environment
ubuntu1~18.04
gcc version 9.4.0 
Synopsys vcs version S-2021.09-SP2-7

**ISSUE**
....../out/build/tb/compile_tb.log

/home/t0252202/RK/ibex_tools/riscv-isa-sim/build/../riscv/interactive.cc:249: undefined \
reference to `htif_t::done()'
/home/t0252202/RK/ibex_tools/lib/libriscv.a(interactive.o): In function `sim_t::wout(boost::asio::basic_streambuf<std::allocator<char> \
>*)':
/usr/include/boost/system/error_code.hpp:439: undefined reference to `boost::system::system_category()' \

/usr/include/boost/system/error_code.hpp:439: undefined reference to `boost::system::system_category()' \

/usr/include/boost/system/error_code.hpp:439: undefined reference to `boost::system::system_category()' \

/home/t0252202/RK/ibex_tools/lib/libriscv.a(interactive.o): In function `sim_t::rin[abi:cxx11](boost::asio::basic_streambuf<std::allocator<char> \
>*)':
/usr/include/boost/system/error_code.hpp:439: undefined reference to `boost::system::system_category()' \

/usr/include/boost/system/error_code.hpp:439: undefined reference to `boost::system::system_category()' \

/home/t0252202/RK/ibex_tools/lib/libriscv.a(interactive.o):/usr/include/boost/asio/error.hpp:230: \
more undefined references to `boost::system::system_category()' follow
/home/t0252202/RK/ibex_tools/lib/libriscv.a(interactive.o): In function `sim_t::rin[abi:cxx11](boost::asio::basic_streambuf<std::allocator<char> \
>*)':
/usr/include/boost/regex/v4/basic_regex.hpp:381: undefined reference to `boost::basic_regex<char, \
boost::regex_traits<char, boost::cpp_regex_traits<char> > >::do_assign(char const*, \
char const*, unsigned int)'
/home/t0252202/RK/ibex_tools/lib/libriscv.a(interactive.o): In function `sim_t::rin[abi:cxx11](boost::asio::basic_streambuf<std::allocator<char> \
>*)':
/usr/include/boost/system/error_code.hpp:439: undefined reference to `boost::system::system_category()' \

this error leading to 
Makefile:117: recipe for target 'product_timestamp' failed

Please let me know,
Am I missing something on setup or configuration !! 

Best Regards
Ranjan

rswarbrick, 2023-06-30:
These messages come from `riscv-isa-sim`, which we call "Spike" locally. Is it possible that you've not got the version that we suggest? I'm looking at the tip of the `ibex_cosim` branch in our riscv-isa-sim clone and I don't see any reference to `htif_t`. What version are you looking at?

RanjanThales, 2023-06-30:
Thanks for the quick reply.
At current build, I have used ibex-cosim-v0.5 (not sure that is stable or not !)
as mentioned in https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex.

However, I am okay to use any recommended version, please let me know !!
or if you can provide any pre-build version, that is also fine at my end !

Thanks.

rswarbrick, 2023-06-30:
Ohh! Now I understand how this is possible! The failure you're seeing is at link time, not compile time. You're right, and the code in question *does* use `htif_t::done()`; but the code doesn't need to include the string "htif_t" to do so. D'oh!

I'm not sure how to fix the error, though. In particular, I think that `htif.cc` should indeed define a symbol called `done()`. I realise that it's a bit of work, but I think you might have to use `objdump` on some compiled objects (maybe `htif.o` and `interactive.o`?) to try to figure out how something has got unstuck. Sorry!

rswarbrick, 2023-06-30:
For what it's worth, I just rebuilt the code in question. There, I see this:
```
rjs@sole:~/work/spike$ objdump -dC htif.o | grep done
0000000000001020 <htif_t::done()>:
rjs@sole:~/work/spike$ objdump -dC interactive.o | grep htif_t
```
In particular, `htif.o` does indeed have an implementation of the `done()` method. It looks like the compilation of `interactive.cc` inlined the definition of `htif_t::done()` (a base class method), which is why it doesn't appear there. I don't really understand what could be different for you, though.

RanjanThales, 2023-07-03:
Hi @rswarbrick, I agree the issue is at link time, not compile time. For trial, I tried to reproduce the same issue #2008 and I could see the exact mentioned error with ibex_cosim-v0.3. (So, at least my steps are correct :))

However the ibex_cosim-v0.5, keep giving the above error.
from the attached compile.log , I doubt some basic library/packages I am missing for this new version (i.e. ibex_cosim-v0.5). 
[config.log](https://github.com/lowRISC/ibex/files/11940571/config.log)
Could you identify, what might be missing at my environment setup  !!
or If you can provide me yours environment details I will try to replicate the same at my end!

rswarbrick, 2023-07-04:
Hmm, I'm really sorry but I don't know. @GregAC, do you have any ideas? (GCC version, maybe?)

RanjanThales, 2023-07-06:
Closing this issue, as this issue got resolved with updated gcc (gcc 11.3) and some OS lib updates.

closed 2023-07-06 by RanjanThales