lowRISC/ibex #2064: Python script Error when running a dv test

liangmin8743, 2023-07-29:
Hi, i'm a beginer of ibex, i run some test to learn ibex dv, and the command didn't work:

make TEST=riscv_machine_mode_rand_test SIMULATOR=vcs

Got the following error:

[image]

Could any one help to solve this problem?

rswarbrick, 2023-07-31:
I'm not entirely sure what's wrong here, but the error message is your Python environment complaining that it doesn't know how do call a `classmethod` object as a function. I suspect this is a Python version problem: if you run `python3 --version`, what gets printed out?

Note that it's worth us (as the authors) getting to the bottom of this: if it turns out that we're missing a more understandable version check, we should add it in! :-)

liangmin8743, 2023-08-01:
> I'm not entirely sure what's wrong here, but the error message is your Python environment complaining that it doesn't know how do call a `classmethod` object as a function. I suspect this is a [...]

python version:  python 3.7.1

In metadata.py,  Line 365:

md = RegressionMetadata.arg_list_initializer(...)

change to 

md = RegressionMetadata().arg_list_initializer(...)

and the script got work...

rswarbrick, 2023-08-02:
That's a bit odd: `RegressionMetadata` is the name of a class, so the solution that you have found is to call the class as a function(!). I don't think we'd be able to take that in our source, but if it works on your machine then I guess that helps! 

I just checked and the official stable release of Python is 3.8, which was released in 2019. We probably won't want to take changes if they are only needed by a version of Python older than the current stable release.

I realise that this isn't a very helpful answer. If you can find some information about what exactly is going wrong, we could definitely tweak our scripts. Even better if you find why it's wrong and how it should be written :-)