lowRISC/ibex #2071: make[1]: *** [out/metadata/tb.compile.stamp] Error 2

cllll402, 2023-08-12:
I am a beginner in ibex, and when I followed the github steps to test something wrong, can you tell me what I missed or did wrong?
The error message is as follows:

autumn@localhost ~/ibex/dv/uvm/core_ibex$ make --keep-going IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0
Building RTL testbench
23:54:56 (snpslmd) OUT: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1691823296] 
23:55:19 (snpslmd) IN: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1691823296] 
23:55:19 (snpslmd) OUT: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1691823319] 
23:55:21 (snpslmd) IN: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1691823319] 
make[1]: *** [out/metadata/tb.compile.stamp] Error 2
Generating core configuration file
Building randomized test generator
23:55:24 (snpslmd) OUT: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1691823324] 
23:55:41 (snpslmd) IN: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1691823324] 
23:55:41 (snpslmd) OUT: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1691823341] 
23:55:43 (snpslmd) IN: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1691823341] 
Running randomized test generator to create assembly file /home/autumn/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/test.S
23:55:44 (snpslmd) OUT: "VCSRuntime_Net" autumn@localhost.localdomain  [snps_checkout_1691823344] 
23:55:51 (snpslmd) IN: "VCSRuntime_Net" autumn@localhost.localdomain  [snps_checkout_1691823344] 
Compiling riscvdv test assembly to create binary at /home/autumn/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/test.bin
make[1]: Target `all' not remade because of errors.
make: *** [run] Error 2

Here are the details of how I use the tools:

VMware：16Pro
Linux：CentOS7.9
VCS：2020.03-SP2-7
Python：3.10.5
Pip：23.2.1
Gcc/G++：4.8.5
Spike：riscv-isa-sim-ibex-cosim-v0.3
Pre-built toolchain：lowrisc-toolchain-gcc-rv32imcb-20230519-1

Finally, some of my path information in bashrc:
#riscv-gcc
export RISCV=/home/autumn/lowrisc-gcc
export RISCV_TOOLCHAIN=$RISCV
export RISCV_GCC=$RISCV/bin/riscv32-unknown-elf-gcc
export RISCV_OBJCOPY=$RISCV/bin/riscv32-unknown-elf-objcopy
export SPIKE_PATH=$RISCV/bin
export PKG_CONFIG_PATH=$RISCV/lib/pkgconfig
export PATH=$PATH:/root/.local/bin #.txt

rswarbrick, 2023-08-14:
Hi there! I'm not sure exactly what has gone wrong, but any failure must have happened just before the line that says ` make[1]: *** [out/metadata/tb.compile.stamp] Error 2`. (Note: the other text that appears afterwards would probably not be there if you hadn't passed `--keep-going`. It's often easier to debug what's going on without that flag)

Anyway, I'm not entirely sure what failed, and we should probably tweak things so that the error messages are clearer. Can you try running again with an extra `VERBOSE=1`. That should hopefully show more about what is going on:
```
make IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0 VERBOSE=1
```

cllll402, 2023-08-14:
First of all, thank you very, very much for replying to my question.
Next I re-output the test as you suggested and look at the end of compile_tb_stdstreams.log. The result is as follows:

autumn@localhost ~/ibex/dv/uvm/core_ibex$ make IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0 VERBOSE=1
Building RTL testbench
env PYTHONPATH=/home/autumn/ibex:/home/autumn/ibex/util:/home/autumn/ibex/dv/uvm/core_ibex/scripts:/home/autumn/ibex/dv/uvm/core_ibex/riscv_dv_extension:/home/autumn/ibex/dv/uvm/core_ibex/yaml:/home/autumn/ibex/vendor/google_riscv-dv/scripts \
scripts/compile_tb.py \
  --dir-metadata out/metadata
+ vcs -full64 -f /home/autumn/ibex/dv/uvm/core_ibex/ibex_dv.f -l /home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb.log -sverilog -ntb_opts uvm-1.2 +define+UVM +define+UVM_REGEX_NO_DPI -timescale=1ns/10ps -licqueue -LDFLAGS -Wl,--no-as-needed -Mdir=/home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv.csrc -o /home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv -debug_access+pp -xlrm uniq_prior_final -CFLAGS '--std=c99 -fno-extended-identifiers' -lca -kdb -pvalue+core_ibex_tb_top.RV32E=0 +define+IBEX_CFG_RV32M=ibex_pkg::RV32MSingleCycle +define+IBEX_CFG_RV32B=ibex_pkg::RV32BOTEarlGrey +define+IBEX_CFG_RegFile=ibex_pkg::RegFileFF -pvalue+core_ibex_tb_top.BranchTargetALU=1 -pvalue+core_ibex_tb_top.WritebackStage=1 -pvalue+core_ibex_tb_top.ICache=1 -pvalue+core_ibex_tb_top.ICacheECC=1 -pvalue+core_ibex_tb_top.ICacheScramble=1 -pvalue+core_ibex_tb_top.BranchPredictor=0 -pvalue+core_ibex_tb_top.DbgTriggerEn=1 -pvalue+core_ibex_tb_top.SecureIbex=1 -pvalue+core_ibex_tb_top.PMPEnable=1 -pvalue+core_ibex_tb_top.PMPGranularity=0 -pvalue+core_ibex_tb_top.PMPNumRegions=16 -pvalue+core_ibex_tb_top.MHPMCounterNum=10 -pvalue+core_ibex_tb_top.MHPMCounterWidth=32 -f /home/autumn/ibex/dv/uvm/core_ibex/ibex_dv_cosim_dpi.f -LDFLAGS -Wl,-rpath,/home/autumn/lowrisc-gcc/lib -CFLAGS '-I/home/autumn/lowrisc-gcc/include -I/home/autumn/lowrisc-gcc/include/softfloat' -CFLAGS -I/home/autumn/ibex/dv/cosim -lriscv -lsoftfloat -ldl -ldisasm -lfdt -lstdc++ >>/home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb_stdstreams.log 2>&1
 7:39:07 (snpslmd) OUT: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1692023947] 
 7:39:42 (snpslmd) IN: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1692023947] 
 7:39:42 (snpslmd) OUT: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1692023982] 
 7:39:54 (snpslmd) IN: "VCSCompiler_Net" autumn@localhost.localdomain  [snps_checkout_1692023982] 
make[1]: *** [out/metadata/tb.compile.stamp] Error 2
make: *** [run] Error 2

All of 94 modules done
make[2]: Entering directory `/home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv.csrc'
make[2]: Leaving directory `/home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv.csrc'
make[2]: Entering directory `/home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv.csrc'
rm -f _cuarc*.so _csrc*.so pre_vcsobj_*.so share_vcsobj_*.so
cc1plus: warning: command line option '-std=c99' is valid for C/ObjC but not for C++ [enabled by default]
g++ -w  -pipe -DVCSMX -DUVM_DPI_DO_TYPE_CHECK -fPIC --std=c99 -fno-extended-identifiers -I/home/autumn/lowrisc-gcc/include -I/home/autumn/lowrisc-gcc/include/softfloat -I/home/autumn/ibex/dv/cosim -O -I/usr/Synopsys/vcs/Q-2020.03-SP2-7/include    -c /usr/Synopsys/vcs/Q-2020.03-SP2-7/etc/uvm-1.2/dpi/uvm_dpi.cc
cc1plus: warning: command line option '-std=c99' is valid for C/ObjC but not for C++ [enabled by default]
g++ -w  -pipe -DVCSMX -DUVM_DPI_DO_TYPE_CHECK -fPIC --std=c99 -fno-extended-identifiers -I/home/autumn/lowrisc-gcc/include -I/home/autumn/lowrisc-gcc/include/softfloat -I/home/autumn/ibex/dv/cosim -O -I/usr/Synopsys/vcs/Q-2020.03-SP2-7/include    -c /usr/Synopsys/vcs/Q-2020.03-SP2-7/etc/uvm-1.2/verdi/dpi/uvm_verdi_dpi.cpp
gcc -w  -pipe -DVCSMX -DUVM_DPI_DO_TYPE_CHECK -fPIC --std=c99 -fno-extended-identifiers -I/home/autumn/lowrisc-gcc/include -I/home/autumn/lowrisc-gcc/include/softfloat -I/home/autumn/ibex/dv/cosim -O -I/usr/Synopsys/vcs/Q-2020.03-SP2-7/include    -c /home/autumn/ibex/dv/uvm/core_ibex/common/date.c
cc1plus: warning: command line option '-std=c99' is valid for C/ObjC but not for C++ [enabled by default]
In file included from /usr/include/c++/4.8.2/cstdint:35:0,
                 from /home/autumn/lowrisc-gcc/include/riscv/decode.h:11,
                 from /home/autumn/lowrisc-gcc/include/riscv/devices.h:4,
                 from /home/autumn/ibex/dv/cosim/spike_cosim.h:16,
                 from /home/autumn/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/spike_cosim_dpi.cc:10:
/usr/include/c++/4.8.2/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
make[3]: *** [spike_cosim_dpi.d] Error 1
make[2]: *** [product_clean_order] Error 2
make[2]: Leaving directory `/home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv.csrc'
Make exited with status 2
CPU time: 28.322 seconds to compile + 1.091 seconds to elab + 1.235 seconds to link
Verdi KDB elaboration done and the database successfully generated: 0 error(s), 0 warning(s)
Running compile_tb command :
[... cut ...] b_top.MHPMCounterWidth=32 -f /home/autumn/ibex/dv/uvm/core_ibex/ibex_dv_cosim_dpi.f -LDFLAGS -Wl,-rpath,/home/autumn/lowrisc-gcc/lib -CFLAGS -I/home/autumn/lowrisc-gcc/include -I/home/autumn/lowrisc-gcc/include/softfloat -CFLAGS -I/home/autumn/ibex/dv/cosim -lriscv -lsoftfloat -ldl -ldisasm -lfdt -lstdc++

For the C++11 mentioned, is it because I used scl enable devtoolset-9 bash when installing spike? Or is it something else? How can I correct it? I am eager to know the answer to this question.

rswarbrick, 2023-08-14:
Well, the first failure happens when running VCS. See line 11, which ends with "Error 2". You're right that some things look a bit dubious below (maybe you're using an old GCC/G++?), but I'd not bother looking too hard at them yet.

I'd expect that you can reproduce the initial failure by copy-pasting and running the first `vcs` command. I think you'll find that it spits some error messages out to `/home/autumn/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb.log`.

cllll402, 2023-08-26:
Thank you, I have solved this problem.
The specific method is to replace ibex-isa-sim-v0.3 with ibex-isa-sim-v0.5, and specify the locations of the five libraries libfdt.a, libriscv.a, libdisasm.a, libsoftfloat.a and libsoftfloat.so. You can solve the problem. Of course, all tools should be updated to newer versions.
The reason for the change of spike is that I found that the 0.3 version of the tool lacks the improvement of mhpmcounter related functions.

cllll402, 2023-08-26:
Next, I tested all the files in test.yaml. Of the 55 projects to be tested, except riscv_bitmanip_full_test, the pass rate of the remaining projects is on schedule toward the official report. https://ibex.reports.lowrisc.org/opentitan/latest/report.html

When I execute the riscv_bitmanip_full_test command alone:
autumn@localhost ~/ibex/dv/uvm/core_ibex$ make --keep-going IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1  SEED=1 TEST=riscv_bitmanip_full_test WAVES=0 COV=0
WARNING:ibex_cmd:Rejecting test riscv_bitmanip_full_test, 'rtl_params' specified not compatible with ibex_config
Traceback (most recent call last):
File "/home/autumn/ibex/dv/uvm/core_ibex/./scripts/metadata.py", line 511, in
sys.exit(_main())
File "/home/autumn/ibex/dv/uvm/core_ibex/./scripts/metadata.py", line 379, in _main
raise RuntimeError("md.tests_and_counts is empty, cannot get TEST.SEED strings.")
RuntimeError: md.tests_and_counts is empty, cannot get TEST.SEED strings.
make: *** [run] Error 1

I found about warnings, found similar problems in https://github.com/lowRISC/ibex/pull/999, but referred to in the file, the code seems to have been corrected, so I have no idea, could you tell me how to do this problem should be?

cllll402, 2023-09-02:
Thanks for your reply, I just want to fully test all the files in test.yaml and have no other purpose. If it's not important, then I would consider not testing it.

rswarbrick, 2023-09-04:
Ok, cool. I guess the remaining question is whether there's a "run everything I need" command that caused `riscv_bitmanip_full_test` to run with `IBEX_CONFIG=opentitan`. If so, there's some filtering we need to fix! What command did you run that caused the test to run in the first place?

xiaoan109, 2025-01-15:
I use  VCS and got this error,
make SIMULATOR=vcs
```
In file included from /home/yian/workspace/ibex/dv/cosim/spike_cosim.cc:15:
/home/yian/tools/spike/include/riscv/mmu.h:13:10: fatal error: byteorder.h: No such \
file or directory
 #include "byteorder.h"
          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [filelist:29: spike_cosim.d] Error 1
make[2]: *** [Makefile:113: product_clean_order] Error 2
Make exited with status 2
```

Can you help me if you met this before?