lowRISC/ibex #2073: Can't run fusesoc properly anymore

Stephane-london, 2023-08-14:
When running "fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast" as given in example, the command get errors when launching the Vibex_simple_system.mk generated file.

In that file, we have those lines, 

.
.
.

# Path to Verilator kit (from $VERILATOR_ROOT)
VERILATOR_ROOT = /sw/freetools/verilator/4.214/rh70_64
.
.
.
# Include global rules
include $(VERILATOR_ROOT)/include/verilated.mk

The include file cannot be found. To me, the root variable should be defined instead as:
VERILATOR_ROOT = /sw/freetools/verilator/4.214/rh70_64/share/verilator

At least, it was the case and correctly running 3 months ago.

Do you face same issues on your side?
Rgds, Stephane.

rswarbrick, 2023-08-14:
Hi there! I think something went a little wrong with posting this, and I'm seeing dots instead of the lines you're trying to show. Could you tweak things to include them properly?

Stephane-london, 2023-08-14:
Hi,
Dot lines in my message are just referring to some lines inside makefile without interests.
Important lines are shown explicitly.
Does it clarify?

rswarbrick, 2023-08-14:
Ah, now I understand. I've added some back-ticks to the original text, which should make it a bit clearer. Thanks.

Stephane-london, 2023-08-14:
many thxs, this is much more clear now!

rswarbrick, 2023-08-14:
Hmm, I think this is caused by your Verilator installation not quite working right. The Makefile line in question comes from Verilator's `V3EmitMk.cpp`, where it looks like it comes from your `VERILATOR_ROOT` environment variable. I suspect that is set to something a bit surprising?

I'm closing the Ibex issue because I think this is a tooling problem, not caused by us :-) Good luck getting everything straight!

closed 2023-08-14 by rswarbrick

Stephane-london, 2023-08-14:
Ok, you may be right. Thank you for your support.
Rgds, S.