lowRISC/ibex #2080: Ask a question related to Ibex

Aaronyap2002, 2023-08-29:
Good evening, I am an undergraduate student doing research. I was trying to simulate the ibex core by following the instructions in the "Examples" directory and i encountered 2 issues. 1) While i was trying to install Vivado 2023.1  in Ubuntu 20.04, but the installation got stuck while "generating device list" 2) After cloning the ibex repo and running the command " pip3 install -U -r python-requirements.txt", i got an error while installing the list of prerequisites in the file (in the picture).  May i know why this happened, and how to fix this ?

**EDA tool and version:**
Xilinx Vivado 2023.1 

**Operating system:**
UBuntu 20.04

**Version of the Ibex source code:**
1eb0beafa5ed23fc0403386bf83367f7c84af669

Aaronyap2002, 2023-08-29:
I ran the command " pip install docutils==0.18.1" Then reran " pip3 install -U -r python-requirements.txt", the error seems to be gone now. But as i ran the command "fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast" to build the simulation, i got the following error ( in the picture). 
[image]

rswarbrick, 2023-08-30:
The error message is Verilator complaining that it doesn't understand the syntax in `prim_cipher.vlt` (to start with). What version of Verilator do you have installed?

If it's older than the minimum listed in our documentation (4.028), no worries! In that case, please close this issue.

Aaronyap2002, 2023-08-30:
Hi, i have checked that i have installed the latest version of verilator
[image]

rswarbrick, 2023-08-30:
I think you want `verilator --version`. Your screenshot shows you typing the words the other way around.

Aaronyap2002, 2023-08-30:
yeah sorry about that, but below that, it shows that i have already installed the latest versio

rswarbrick, 2023-08-30:
Ahah! The problem is that we're using `split_var` in the code, which was only in Verilator 4.030 (I just found it in Verilator's changelog).

To get things working for you, please jump to a recent version of Verilator. To close this issue, I'll push a change to the required version in a minute.

Thank you very much for the bug report.

Aaronyap2002, 2023-08-30:
May i know how do i install 4.03? I couldn't find it

rswarbrick, 2023-08-30:
About installing a more recent version of the tool: it looks like Ubuntu 20.04 doesn't have a recent Verilator packaged. It's probably best to follow the "Git Quick Install" section here: https://verilator.org/guide/latest/install.html.

Aaronyap2002, 2023-08-30:
i installed version 5.014, and it produced different kind of error
[image]

rswarbrick, 2023-08-30:
Hmm, yes. It's annoying! I've been trying (and failing) to come up with a syntax for that file that is acceptable to various versions of Verilator, and Verible, and VCS. As a quick hack, I'd suggest adding the `lint_off` line that it's suggesting. If you'd be willing to make a PR from the result, that would be wonderful.

thishakyabb, 2023-08-30:
I was able to bypass the after using the lint_off. Thank you very much for that.  When Im trying to build the software using make -C command I got this error. I think I may have placed the pre built gcc based toolchain somewhere wrong. I tried to find how to place and use the toolchain for hours but I could't. Could you please tell me where should I place the toolchain and  If there's things that I should do additionally . 
Thank you[image]

closed 2023-08-31 by rswarbrick

Aaronyap2002, 2023-08-31:
May I know which version of the verilator should I add the lint_off command? And which command should I add it? Thank you so much!

PR #2081 (2023-08-31): [doc] Bump minimum Verilator version
It turns out that we use split_var in the code, which was added in version 4.030 (and several bugs were fixed in following versions). Change the minimum required version to match what we're using in CI (and presumably works!)

Closes #2080.