lowRISC/ibex #2085: IBEX Compiling issue

Krishnakumarmohanraj, 2023-09-13:
make IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0 VERBOSE=1

Getting error as make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 2
make: *** [Makefile:66: run] Error 2

When i check log file out/build/tb/compile_tb.log

it showing

/tools/spike-11.09.2023/include/riscv/csrs.h:843:8: error: 'optional' in namespace \
'std' does not name a template type
  843 |   std::optional<reg_t> prev_val;
      |        ^~~~~~~~
/tools/spike-11.09.2023/include/riscv/csrs.h:843:3: note: 'std::optional' is only \
available from C++17 onwards
  843 |   std::optional<reg_t> prev_val;
      |   ^~~
In file included from /tools/spike-11.09.2023/include/riscv/processor.h:17,
                 from /Projects/marmik_project/manicka.vinodhini/TRAIN/ibex/dv/cosim/spike_cosim.h:18, \

OS: Centos 7
GCC- Devtoolset-9 (gcc (GCC) 9.3.1 20200408 (Red Hat 9.3.1-2))
Python 3.7.0
Spike RISC-V ISA Simulator 1.1.1-dev
Spike lib files as below
libcustomext.so  libfdt.so   libriscv.so      pkgconfig
libdisasm.a      libfesvr.a  libsoftfloat.so

rswarbrick, 2023-09-14:
This is an error building the Spike code (a dependency that we don't maintain). It looks like you've got a recent version of upstream's Spike: that code was added in June this year.

To get things working, you'll need the version of spike that we describe in the documentation.

closed 2023-09-14 by rswarbrick

Krishnakumarmohanraj, 2023-09-15:
Hi Mr.Rupert Swarbrick,

 I downloaded exact spike version "riscv-isa-sim-ibex-cosim-v0.5.tar.gz" and configure with --enable-commitlog --enable-misaligned option, then when we run "make IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0 VERBOSE=1"

am getting following error
/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/ld: cannot find -lriscv \

/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/ld: cannot find -lsoftfloat \

/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/ld: cannot find -ldisasm \

/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/ld: cannot find -lfdt \

When we checked the spike installation location we got .a file and not getting .so file, I pasted the files detail as below
lib]$ ls
libcustomext.so  libfdt.a    libriscv.a       pkgconfig
libdisasm.a      libfesvr.a  libsoftfloat.so

phkconfig folder having:
riscv-disasm.pc  riscv-fdt.pc  riscv-fesvr.pc  riscv-riscv.pc

Any suggestion on this please?

rswarbrick, 2023-09-15:
That file list looks sensible to me. I suspect that the build command isn't passing the right `-L` flags to the linker. Maybe it's worth you looking at the `ld` command that actually ran. I think we pick up those flags from `pkgconfig`, so you might find you need to check that you have set up `PKG_CONFIG_PATH` to something sensible.

Krishnakumarmohanraj, 2023-09-16:
Hi Mr.Rupert Swarbrick,

    **if we run make without PFG_CONFIG_PATH, am getting below error**

Traceback (most recent call last):
  File "scripts/compile_tb.py", line 75, in _main
    subprocess.check_output(['pkg-config', '--exists'] + spike_iss_pc)
  File "/opt/python-3.7.0/lib/python3.7/subprocess.py", line 376, in check_output
    **kwargs).stdout
  File "/opt/python-3.7.0/lib/python3.7/subprocess.py", line 468, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['pkg-config', '--exists', 'riscv-riscv', 'riscv-disasm', 'riscv-fdt']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "scripts/compile_tb.py", line 110, in <module>
    sys.exit(_main())
  File "scripts/compile_tb.py", line 79, in _main
    f'Did you set the PKG_CONFIG_PATH correctly?') from err
RuntimeError: Failed to find ['riscv-riscv', 'riscv-disasm', 'riscv-fdt'] pkg-config packages. Did you set the PKG_CONFIG_PATH correctly?

**But if I set PKG_CONFIG_PATH, setenv PKG_CONFIG_PATH /Projects/marmik_project/manicka.vinodhini/krishna/spike-13.09.23/lib/pkgconfig. am getting error as below**

/tools/VCS_VT_2022.06.SP1/vcs/T-2022.06-SP1/linux64/lib/vcs_save_restore_new.o -ldl \
-lc -lm -lpthread -ldl
/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/ld: cannot find -lriscv \

/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/ld: cannot find -lsoftfloat \

/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/ld: cannot find -ldisasm \

/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/ld: cannot find -lfdt \

collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:117: product_timestamp] Error 1

rswarbrick, 2023-09-18:
OK, looks like you have set PKG_CONFIG_PATH. You'll still need to look at the actual `ld` command to work out what's going wrong. This is also now a little beyond what you might expect as help from a GitHub issue conversation. :-) Good luck getting things working!

Krishnakumarmohanraj, 2023-09-29:
Hi Rupert Swarbrick, 

     we installed ubuntu and installed spike and ibex , when we run "make IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0 VERBOSE=1"  am getting below error

[image]
[image]

Let me know which gcc version we need to use in ubuntu 22.04