lowRISC/ibex #2091: Building Simulation error

lipf1987, 2023-10-02:
Building Simulation error：
fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast 
INFO: Preparing lowrisc:dv:crypto_prince_ref:0.1
INFO: Preparing lowrisc:dv:dv_fcov_macros:0
INFO: Preparing lowrisc:dv:secded_enc:0
INFO: Preparing lowrisc:dv_verilator:ibex_pcounts:0
INFO: Preparing lowrisc:dv_verilator:simutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_pkg:0.1
INFO: Preparing lowrisc:prim:primgen:0.1
INFO: Preparing lowrisc:prim:ram_1p_pkg:0
INFO: Preparing lowrisc:prim:ram_2p_pkg:0
INFO: Preparing lowrisc:prim:util_get_scramble_params:0
INFO: Preparing lowrisc:tool:check_tool_requirements:0.1
INFO: Preparing lowrisc:dv:scramble_model:0
INFO: Preparing lowrisc:dv_verilator:memutil_dpi:0
INFO: Preparing lowrisc:lint:common:0.1
INFO: Preparing lowrisc:prim:prim_pkg:0.1
INFO: Preparing lowrisc:dv_verilator:memutil_dpi_scrambled:0
INFO: Preparing lowrisc:dv_verilator:memutil_verilator:0
INFO: Preparing lowrisc:prim:assert:0.1
INFO: Preparing lowrisc:prim:cipher_pkg:0.1
INFO: Preparing lowrisc:prim:clock_gating:0
INFO: Preparing lowrisc:prim:clock_mux2:0
INFO: Preparing lowrisc:prim:ram_1p:0
INFO: Preparing lowrisc:prim:ram_2p:0
INFO: Preparing lowrisc:prim:secded:0.1
INFO: Preparing lowrisc:ibex:ibex_icache:0.1
INFO: Preparing lowrisc:ibex:ibex_tracer:0.1
INFO: Preparing lowrisc:ibex:sim_shared:0
INFO: Preparing lowrisc:prim:buf:0
INFO: Preparing lowrisc:prim:cipher:0
INFO: Preparing lowrisc:prim:flop:0
INFO: Preparing lowrisc:prim:lfsr:0.1
INFO: Preparing lowrisc:prim:util:0.1
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:prim:onehot_check:0
INFO: Preparing lowrisc:prim:ram_1p_adv:0.1
INFO: Preparing lowrisc:prim:ram_1p_scr:0.1
INFO: Preparing lowrisc:ibex:ibex_top:0.1
INFO: Preparing lowrisc:ibex:ibex_top_tracing:0.1
INFO: Preparing lowrisc:ibex:ibex_simple_system_core:0
INFO: Preparing lowrisc:ibex:ibex_simple_system:0
INFO: Generating lowrisc:prim:prim_pkg-impl:0.1
Creating prim_pkg.sv
Core file written to prim_pkg.core.
INFO: Generating lowrisc:prim:clock_gating-impl:0
Implementations for primitive clock_gating: xilinx, generic
Inspecting generic module /home/lipengfei2/disk/tmp/Ibex/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_clock_gating.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/lipengfei2/disk/tmp/Ibex/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.sv
Creating core file for primitive clock_gating.
Core file written to /home/lipengfei2/disk/tmp/Ibex/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.core
INFO: Generating lowrisc:prim:clock_mux2-impl:0
Implementations for primitive clock_mux2: generic, xilinx
Inspecting generic module /home/lipengfei2/disk/tmp/Ibex/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_clock_mux2.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/lipengfei2/disk/tmp/Ibex/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_mux2-impl_0/prim_clock_mux2.sv
Creating core file for primitive clock_mux2.
Core file written to /home/lipengfei2/disk/tmp/Ibex/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_mux2-impl_0/prim_clock_mux2.core
INFO: Generating lowrisc:prim:ram_1p-impl:0
Implementations for primitive ram_1p: generic, badbit
Inspecting generic module /home/lipengfei2/disk/tmp/Ibex/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_ram_1p.sv
No module named 'anytree'
Verible parser failed, using regex fallback instead.
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/lipengfei2/disk/tmp/Ibex/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_1p-impl_0/prim_ram_1p.sv
[... 51 lines cut ...]
  763 |     data_accessed = 5'h0;
      |     ^~~~~~~~~~~~~
                      ../src/lowrisc_ibex_ibex_tracer_0.1/rtl/ibex_tracer.sv:702:7: ... Location of other write
  702 |       data_accessed = RS1 | RS2 | MEM;
      |       ^~~~~~~~~~~~~
%Warning-BLKSEQ: ../src/lowrisc_ibex_ibex_tracer_0.1/rtl/ibex_tracer.sv:119:19: Blocking assignment '=' in sequential logic process
                                                                              : ... In instance ibex_simple_system
                                                                              : ... Suggest using delayed assignment '<='
  119 |       file_handle = $fopen(file_name, "w");
      |                   ^
%Error: Exiting due to 3 warning(s)
make: *** [Vibex_simple_system.mk] Error 1

ERROR: Failed to build lowrisc:ibex:ibex_simple_system:0 : '['make', '-j', '16']' exited with an error: 2

MY OS:
NAME="Ubuntu"
VERSION="18.04.4 LTS (Bionic Beaver)"

Verilator 5.016 2023-09-16 rev v5.014-149-g57c816f90

ibex's git log
dccad9e6a39b196261e1ba4586eb8a1711783fed

Thanks for your help

rswarbrick, 2023-10-03:
Ah, that makes sense. This is the same problem as we had to fix for OTBN in https://github.com/lowRISC/opentitan/pull/17971.

This was a bit easier for OTBN. To get things working in Ibex, a problem is that the decode functions (e.g. `decode_r_insn`) fundamentally need to produce both `decoded_str` and `data_accessed`.

Clearly, we could silence the warnings with the `lint_off` lines suggested in the error message.

Alternatively, we could tweak our decode functions slightly so that they return the string and bitmask as a struct. I *think* that would probably work reasonably well.

@lipf1987: Would you be happy to try one of these approaches and make a PR? (@GregAC: Do they make sense to you? Anything simpler that we could do?)

rswarbrick, 2023-10-03:
We still need to fix it properly at our end :-) I'll push a hacky PR for now and tag this issue with it.

PR #2093 (2023-11-22): Fix a couple of Verilator warnings
This waives/fixes a couple of warnings that appear with recent versions of Verilator. See the commit messages for details.

Fixes #2091.
commit 20183012 (2023-11-22): [verilator] Waive MULTIDRIVEN warning in ibex_tracer.sv

Fixes #2091.
closed 2023-11-22 by rswarbrick