lowRISC/ibex #2095: python flow error:  dv/uvm/core_ibex/scripts/riscvdv_interface.py", line 165, in get_tool_cmds     raise RuntimeError

A-B-S-Anik, 2023-10-18:
I am trying to learn ibex verification flow.
After following the instruction correctly, I am facing a runtime error.
I don't know what I am doing wrong.
This is the command I used:   make SIMULATOR=questa
Please Help!
I saw a similar problem in here: https://github.com/lowRISC/ibex/issues/1826
TIA

Build metadata already exists, not recreating from scratch.
Building RTL testbench
Parameters in riscvdv command not substituted!
Parameters : ['<rtl_log>']
Command :  vlog
  -64
  -f /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/ibex_dv.f
  -sv
  -mfcu -cuname design_cuname
  +define+UVM_REGEX_NO_DPI
  +define+UVM
  -timescale \"1 ns / 1 ps \"
  -writetoplevels /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/top.list
  -l /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/<rtl_log>
  +define+IBEX_CFG_RV32M=ibex_pkg::RV32MSingleCycle +define+IBEX_CFG_RV32B=ibex_pkg::RV32BOTEarlGrey +define+IBEX_CFG_RegFile=ibex_pkg::RegFileFF

Traceback (most recent call last):
  File "/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 110, in <module>
    sys.exit(_main())
  File "/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 87, in _main
    md.tb_build_cmds = riscvdv_interface.get_tool_cmds(
  File "/home/siddek/.local/lib/python3.10/site-packages/typeguard/__init__.py", line 1033, in wrapper
    retval = func(*args, **kwargs)
  File "/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/scripts/riscvdv_interface.py", line 165, in get_tool_cmds
    raise RuntimeError
RuntimeError
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 1
make: *** [Makefile:66: run] Error 2

## My Environment

**EDA tool and version:**
Questa Sim-64 2021,4
Latest Spike RISC-V ISA Simulator 1.1.1-dev from ibex
Latest Prebuild riscv tollchain version 10.2.0
Latest RISCV-DV 71666ebacd69266b1abb7cdbad5e1897ce5884e6

**Operating system:**

Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04

**Version of the Ibex source code:**
97c0a7231a39f1be13f04bb37529d95e126f3f43

hcallahan-lowrisc, 2023-10-18:
Hi @A-B-S-Anik, thanks for creating a detailed issue with your problem.
The issue here is that we don't actively use or test ibex verification with questa, so the flow unfortunately got broken at some point. I can certainly help get it going again with feedback about any problems you encounter. I don't think it's too far away from working. However, I can't run a simulation here to test the full flow!
I created a branch here with some changes which I think should get you further. Could you please try it out? https://github.com/lowRISC/ibex/pull/2096

A-B-S-Anik, 2023-10-19:
Hi @hcallahan-lowrisc,
Thank for quick reply. Now there is a timestamp issue.

$ make SIMULATOR=questa VERBOSE=1
WARNING:ibex_cmd:Rejecting test: riscv_bitmanip_full_test. It specifies rtl_params of ['ibex_pkg::RV32BFull'], which doesn't contain the expected 'ibex_pkg::RV32BOTEarlGrey'.
Building RTL testbench
env PYTHONPATH=/home/siddek/work/cosimibex/ibex_new/ibex:/home/siddek/work/cosimibex/ibex_new/ibex/util:/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/scripts:/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/riscv_dv_extension:/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/yaml:/home/siddek/work/cosimibex/ibex_new/ibex/vendor/google_riscv-dv/scripts \
scripts/compile_tb.py \
  --dir-metadata out/metadata
+ vmap mtiUvm '$QUESTA_HOME/questasim/uvm-1.2' >>/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb_stdstreams.log 2>&1
+ vlog -64 -f /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/ibex_dv.f -sv -mfcu -cuname design_cuname +define+UVM_REGEX_NO_DPI +define+UVM -timescale '"1' ns / 1 ps '"' -writetoplevels /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/top.list -l /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb.log +define+IBEX_CFG_RV32M=ibex_pkg::RV32MSingleCycle +define+IBEX_CFG_RV32B=ibex_pkg::RV32BOTEarlGrey +define+IBEX_CFG_RegFile=ibex_pkg::RegFileFF >>/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb_stdstreams.log 2>&1
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 1
make: *** [Makefile:66: run] Error 2

in compile_tb_stdstreams.log:

QuestaSim-64 vmap 2021.4 Lib Mapping Utility 2021.10 Oct 13 2021
vmap mtiUvm $QUESTA_HOME/questasim/uvm-1.2 
Modifying modelsim.ini
QuestaSim-64 vlog 2021.4 Compiler 2021.10 Oct 13 2021
** Error: (vlog-2102) Bad specification for -timescale option.
Errors: 1, Warnings: 0
Running compile_tb command :
vmap mtiUvm $QUESTA_HOME/questasim/uvm-1.2
Running compile_tb command :
vlog -64 -f /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/ibex_dv.f -sv -mfcu -cuname design_cuname +define+UVM_REGEX_NO_DPI +define+UVM -timescale "1 ns / 1 ps " -writetoplevels /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/top.list -l /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb.log +define+IBEX_CFG_RV32M=ibex_pkg::RV32MSingleCycle +define+IBEX_CFG_RV32B=ibex_pkg::RV32BOTEarlGrey +define+IBEX_CFG_RegFile=ibex_pkg::RegFileFF

in compile_tb.log:

QuestaSim-64 vlog 2021.4 Compiler 2021.10 Oct 13 2021
** Error: (vlog-2102) Bad specification for -timescale option.
Errors: 1, Warnings: 0

A-B-S-Anik, 2023-10-19:
> The issue here is that we don't actively use or test ibex verification with questa, so the flow unfortunately got broken at some point

One more question, Ibex verification flow work best with which simulator? Can you please tell me.
TIA

hcallahan-lowrisc, 2023-10-20:
> > The issue here is that we don't actively use or test ibex verification with questa, so the flow unfortunately got broken at some point
> 
> One more question, Ibex verification flow work best with which simulator? Can you please tell me. TIA

Right now, the flow works best with Xcelium and VCS, but there is no reason it couldn't work well with the other tools in `rtl_simulation.yaml`. Those just happen to be the tools we have access to right now for testing and development.

A-B-S-Anik, 2023-10-20:
Hi,
Thank you for the reply.
Different type of error has occurred it's regarding some variables are not properly defined.
Please have a look:

$ make SIMULATOR=questa VERBOSE=1
WARNING:ibex_cmd:Rejecting test: riscv_bitmanip_full_test. It specifies rtl_params of ['ibex_pkg::RV32BFull'], which doesn't contain the expected 'ibex_pkg::RV32BOTEarlGrey'.
Building RTL testbench
env PYTHONPATH=/home/siddek/work/cosimibex/ibex_new/ibex:/home/siddek/work/cosimibex/ibex_new/ibex/util:/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/scripts:/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/riscv_dv_extension:/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/yaml:/home/siddek/work/cosimibex/ibex_new/ibex/vendor/google_riscv-dv/scripts \
scripts/compile_tb.py \
  --dir-metadata out/metadata
+ vmap mtiUvm '$QUESTA_HOME/questasim/uvm-1.2' >>/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb_stdstreams.log 2>&1
+ vlog -64 -f /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/ibex_dv.f -sv -mfcu -cuname design_cuname +define+UVM_REGEX_NO_DPI +define+UVM -timescale 1ns/1ps -writetoplevels /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/top.list -l /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb.log +define+IBEX_CFG_RV32M=ibex_pkg::RV32MSingleCycle +define+IBEX_CFG_RV32B=ibex_pkg::RV32BOTEarlGrey +define+IBEX_CFG_RegFile=ibex_pkg::RegFileFF >>/home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/out/build/tb/compile_tb_stdstreams.log 2>&1
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 2
make: *** [Makefile:66: run] Error 2

in compile_tb_stdstreams.log:

** Warning: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/tests/core_ibex_new_seq_lib.sv(56): (vlog-2240) Treating stand-alone use of function 'randomize' as an implicit VOID cast.
-- Compiling interface core_ibex_fcov_if
** Error: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/fcov/core_ibex_fcov_if.sv(645): (vlog-13097) Could not find coverpoint or cross 'cs_registers_i.mstatus_q.mie' in local scope. Please provide a valid Coverpoint variable.
-- Compiling module core_ibex_fcov_bind
-- Compiling interface core_ibex_pmp_fcov_if
** Warning: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/fcov/core_ibex_pmp_fcov_if.sv(18): (vlog-13314) Defaulting port 'csr_pmp_cfg' kind to 'var' rather than 'wire' due to default compile option setting of -svinputport=relaxed.
** Warning: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/fcov/core_ibex_pmp_fcov_if.sv(19): (vlog-13314) Defaulting port 'csr_pmp_addr' kind to 'var' rather than 'wire' due to default compile option setting of -svinputport=relaxed.
** Warning: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/fcov/core_ibex_pmp_fcov_if.sv(20): (vlog-13314) Defaulting port 'pmp_req_err' kind to 'var' rather than 'wire' due to default compile option setting of -svinputport=relaxed.
** Warning: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/fcov/core_ibex_pmp_fcov_if.sv(21): (vlog-13314) Defaulting port 'csr_pmp_mseccfg' kind to 'var' rather than 'wire' due to default compile option setting of -svinputport=relaxed.
** Error: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/fcov/core_ibex_pmp_fcov_if.sv(677): (vlog-13097) Could not find coverpoint or cross 'cs_registers_i.mstatus_q.mpp' in local scope. Please provide a valid Coverpoint variable.
** Error: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/fcov/core_ibex_pmp_fcov_if.sv(679): (vlog-13097) Could not find coverpoint or cross 'cs_registers_i.mstatus_q.mpp' in local scope. Please provide a valid Coverpoint variable.
** Error: /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/fcov/core_ibex_pmp_fcov_if.sv(690): (vlog-13097) Could not find coverpoint or cross 'cs_registers_i.mstatus_q.mpp' in local scope. Please provide a valid Coverpoint variable.
-- Compiling module core_ibex_tb_top
-- Importing package core_ibex_test_pkg
-- Compiling DPI/PLI C file /home/siddek/work/cosimibex/ibex_new/ibex/dv/uvm/core_ibex/common/date.c
End time: 12:47:26 on Oct 20,2023, Elapsed time: 0:00:01
Errors: 4, Warnings: 70

rayamsumanth, 2023-11-08:
Hi,

Any update on this issue? I am facing the same issue on the mentioned `questa_sim_fixes` branch

bekbeis, 2024-07-03:
Hi @A-B-S-Anik,

Did you figure it out? I am facing the same issue right now.

VeriMaverick, 2024-08-06:
Hello! Can you share the installation package of the software