lowRISC/ibex #2100: Ask a question related to Ibex

twyayaya, 2023-11-15:
Hello, I'm trying to run synthesis with Vivado, and it shows some error messages like these:
```
[Synth 8-1766] cannot open include file prim_assert_dummy_macros.svh ["../ibex_rtl/prim_assert.sv":96]
[Synth 8-2841] use of undefined macro ASSERT ["../ibex_rtl/ibex_branch_predict.sv":91]
[Synth 8-2715] syntax error near ( ["../ibex_rtl/ibex_branch_predict.sv":91]

```

I also tried to set the file type, letting prim_assert_dummy_macros.svh become a "Global Include" file. After changing it, there are still some error messages:
```
[Synth 8-2379] near = `ASSERT_DEFAULT_RST : illegal macro parameter ["../ibex_rtl/prim_assert_dummy_macros.svh":12]
[Synth 8-2379] near = `ASSERT_DEFAULT_RST : illegal macro parameter ["../ibex_rtl/prim_assert_dummy_macros.svh":13]
[Synth 8-2379] near = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST : illegal macro parameter ["../ibex_rtl/prim_assert_dummy_macros.svh":14]
[Synth 8-2379] near = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST : illegal macro parameter ["../ibex_rtl/prim_assert_dummy_macros.svh":15]
[Synth 8-2379] near = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST : illegal macro parameter ["../ibex_rtl/prim_assert_dummy_macros.svh":16]
[Synth 8-2379] near = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST : illegal macro parameter ["../ibex_rtl/prim_assert_dummy_macros.svh":17]
[Synth 8-2379] near = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST : illegal macro parameter ["../ibex_rtl/prim_assert_dummy_macros.svh":18]
```

How can I make the synthesis work?

## My Environment

**EDA tool and version:**

Xilinx Vivado 2020.2

**Operating system:**

Ubuntu Linux 18.04

**Version of the Ibex source code:**

rswarbrick, 2023-11-20:
Hi there! Something slightly odd has happened here: I think the `prim_assert.sv` file should include `prim_assert_standard_macros.svh` with Vivado (not the "dummy" file). Can you send a follow-up with the version of Ibex that you're using? (what happens when you run `git rev-parse HEAD`?) The version that I've got on my machine doesn't have a relevant include on line 96.

Also, it might make sense to dig around in the ifdef list in `prim_assert.sv` and add "debug errors" to make sure you can trace why the code building in the way that gets confused.

virajwadhwa, 2024-04-11:
Hi! I am trying to do the same with Vivado, and cannot find these macro files in the ibex repository. Why would they not be included? I found them in OpenTitan, but I'm not sure if these are the same that are required, and how many I would need to include?

joaofoes, 2024-11-11:
Hello, is there any update on this errors? I'm facing the same issue when trying to bring the lowrisc/ibex core to vivado.