lowRISC/ibex #2107: RISCV-DV not running on RHEL 8. Python files throwing pydantic package error

rayamsumanth, 2023-11-27:
## Observed Behavior

When running the Makefiles in` ${IBEX_ROOT}/dv/uvm/core_ibex`, the python package immediately throws an error:
```
$  make IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0

Traceback (most recent call last):
  File "./scripts/metadata.py", line 32, in <module>
    import directed_test_schema
  File "/home/redefine/Workspace/sumanth/ibex/dv/uvm/core_ibex/scripts/directed_test_schema.py", line 37, in <module>
    class DConfig(pydantic.BaseModel):  # noqa
  File "/home/redefine/Workspace/sumanth/ibex/dv/uvm/core_ibex/scripts/directed_test_schema.py", line 65, in DConfig
    @pydantic.field_validator('ld_script', 'includes', mode='before')
AttributeError: module 'pydantic' has no attribute 'field_validator'

```

- The file path where `@pydantic.field_validator()` is used: `${IBEX_ROOT}/dv/uvm/core_ibex/scripts/directed_test_schema.py`
- Tried to run on Ubuntu with QuestaSim license. Then I am facing this error which is not resolved [[here](https://github.com/lowRISC/ibex/issues/2095)]. 
- Tried running it on VCS. Had to switch to RHEL as there is no VCS support for Ubuntu. Raised a question regarding this [[here](https://github.com/lowRISC/ibex/issues/2106)]. I am facing this issue since then

Been here for at least a good 15-20 days with no improvements. Kindly help out.

## Expected Behavior

The expected behavior is to run the Makefile and generate an output directory for the testcase mentioned

## Steps to reproduce the issue

From ${IBEX_ROOT}/dv/uvm/core_ibex, please execute:
```
 make IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0
```
This is the only command that is run

## My Environment

OS  =  Red Hat Enterprise Linux 8.9
Python version = 3.8.17
Pip version  =  19.3.1
RTL Simulator = VCS

**EDA tool and version:**

Synopsys VCS U-2023.03

**Operating system:**

Red Hat Enterprise Linux 8.9

**Version of the Ibex source code:**

97c0a7231a39f1be13f04bb37529d95e126f3f43

rswarbrick, 2023-11-27:
Hmm, that's not great! I suspect this is a Pydantic versioning problem: they had a backwards-incompatible change that we only switched to using back in August. It might be that you've got the version that we used to require. If that is the problem, I think that `pip3 install -U pydantic` will get it working.

rayamsumanth, 2023-11-29:
I don't know what's weird. The pydantic package was fine. Python version was the issue. Python could not pick up Pydantic 2.5.2 and was defaulting to Pydantic 1.9. After creating a symlink to redirect all my versions to python 3.8 installed directory, the files were taking the correct path. I can close the issue now. In the python_requiements.txt, there may be a need to add a version check >=3.8.

I was able to run basic-arithmetic-test as a sample and it passed the checks.

Thank you for the answer.