lowRISC/ibex #2111: The testcase files

VivianDa, 2023-12-14:
I have sucessfully run the verification tests.
When viewing the UVM testbench for verification of the Ibex core, I know that the complete test list can be found in the file [dv/uvm/core_ibex/riscv_dv_extension/testlist.yaml](https://github.com/lowRISC/ibex/blob/master/dv/uvm/core_ibex/riscv_dv_extension/testlist.yaml). And as the information I get, testbench use RISCV-DV to generate binaries from system verilog. 
But I'm wondering **where is the testcase files? for example riscv_arithmetic_basic_test?**

rswarbrick, 2023-12-14:
Hi there, and thanks for the question.

Many of the tests work in essentially the same way:
- We generate a constrained random program using riscv-dv.
- We execute the program with an instruction set simulator (Spike), spitting out a log of the instructions and state changes.
- We execute the program in an RTL simulation, spitting out equivalent logs.
- We compare the logs from the two executions and check that they match.

Some specific tests / configurations are defined in `testlist.yaml`. For example, it contains:
```
- test: riscv_arithmetic_basic_test
  description: >
    Arithmetic instruction test, no load/store/branch instructions
  gen_opts: >
    +instr_cnt=10000
    +num_of_sub_program=0
    +no_fence=1
    +no_data_page=1
    +no_branch_jump=1
    +boot_mode=m
  iterations: 10
  gen_test: riscv_instr_base_test
  rtl_test: core_ibex_base_test
```

There are some more details about how this all works in the manual. See the file `verification.rst`. You can find it rendered online here: https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html

closed 2023-12-14 by rswarbrick

VivianDa, 2023-12-14:
May i say that when using `make TEST=riscv_arithmetic_basic_test` , in riscv-dv is actually using the three files below to generate tmp *.S file ?
```    
inst_overrides = [
        'riscv_asm_program_gen',
        'ibex_asm_program_gen',
        'uvm_test_top.asm_gen'
    ]
   ...

    sim_opts_dict = {
        'uvm_set_inst_override': ','.join(inst_overrides),
        ...
    }
    ```
 (Code from [dv/uvm/core_ibex/scripts/run_instr_gen.py](https://github.com/lowRISC/ibex/blob/master/dv/uvm/core_ibex/scripts/run_instr_gen.py) line 156 to 175.)