lowRISC/ibex #2116: Is it possible to use newer versions of Verilator

Aaronyap2002, 2023-12-19:
The version suggested in this repository is 4.104. Is it possible to use version 4.210, or even 5.107? The reason is because i would like to build the docker container according to the instruction in `ibex-demo-system`, but the version of verilator version stated in `container/Dockerfile` mentioned 
```
# Global configuration options.
ARG VERILATOR_VERSION=4.210
ARG VERIBLE_VERSION=v0.0-2135-gb534c1fe
ARG RISCV_TOOLCHAIN_RELEASE_TAG=20220210-1
```
I am worried that if i update to the newest version, i might not be able to build the simulation.

Aaronyap2002, 2023-12-19:
i checked my version is 0.1. I can't remember how i instalkled my fusesoc, but yeah i got the 0.1 instead

utku1998, 2023-12-19:
Now, I have tried with all of the versions listed here: "1.4, 1.5, 1.6, 1.6.1, 1.7, 1.8, 1.8.1, 1.8.2, 1.8.3, 1.8.4, 1.9, 1.9.1, 1.9.2, 1.9.3, 1.10, 1.11.0, 1.12.0, 2.0rc1, 2.0, 2.1, 2.2, 2.2.1, 2.3" and it worked for ```2.0```. I checked the version and 2.0 was there.

But then when I try to build the system again, I got the following error: "Key 'cores' not found in GAPI structure. Install a compatible version with 'pip3 install --user -r python-requirements.txt'."" Once I installed the requirements, I could the build the system. But now, fusesoc --version command prints "0.1".

I did how understand how it worked, but it worked, although there is no version like "0.1" listed there.

Aaronyap2002, 2023-12-19:
That's interesting! Have you tried using different versions of verilator as well? Is it possible to share the commands you've used to install different versions of FuseSOC, i would like to replicate it myself too.

utku1998, 2023-12-19:
I did not tried it. Currently, I am using the version "Verilator 5.018 2023-10-30 rev v5.018-38-g344f87abe".

The commands are:
1. Uninstall the Current Version: pip uninstall fusesoc
2. Install the Desired Version: pip install fusesoc==2.0
3. Verify the Installation: fusesoc --version

Please let me know, if you learn something new!

NovaWu, 2024-03-21:
> I have updated the fusesoc to the latest version (2.3), but now I cannot build the simulation, although it works fine before. Now I get the following error:
> 
> """ ... Error validating data.filesets.files_verilator_waiver.files must be array WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_and2.core: Error validating data.filesets.files_verilator_waiver.files must be array ERROR: Conflicting requirements: Requirements: 'lowrisc_ibex_ibex_simple_system == 0-0' <- 'lowrisc_ibex_ibex_simple_system_core >= 0-0' <- 'lowrisc_ibex_sim_shared >= 0-0' <- 'lowrisc_prim_ram_1p >= 0-0' +lowrisc_ibex_sim_shared-0-0 was ignored because it depends on missing packages Requirements: 'lowrisc_ibex_ibex_simple_system == 0-0' <- 'lowrisc_ibex_ibex_simple_system_core >= 0-0' <- 'lowrisc_ibex_sim_shared >= 0-0' lowrisc_ibex_ibex_simple_system_core-0-0 requires (+lowrisc_ibex_sim_shared-0-0) Requirements: 'lowrisc_ibex_ibex_simple_system == 0-0' <- 'lowrisc_ibex_ibex_simple_system_core >= 0-0' lowrisc_ibex_ibex_simple_system-0-0 requires (+lowrisc_ibex_ibex_simple_system_core-0-0) Requirements: 'lowrisc_ibex_ibex_simple_system == 0-0' Install command rule (+lowrisc_ibex_ibex_simple_system-0-0)
> 
> Failed to resolve dependencies for lowrisc:ibex:ibex_simple_system:0 """
> 
> Is there a solution for that? And what version should I use? Since, I don't remember what was the version that works, I can't build the simulation again!

——————————————————————————
I also encountered the same problem, please tell me how you solved it?

thepalbi, 2024-05-09:
I was able to build correctly simple_system, following the instructions [here](https://github.com/lowRISC/ibex/blob/master/examples/simple_system/README.md). First, I got this same error when trying to use `fusesoc` installed through my package manger (2.3), and then checking how the CI installs it, found that inside the python requirements, a fork of fusesoc is used. 

I'd recommend  creating a python virtualenv, installing the requirements there, and using the fusesoc version shipped there:

```bash
python -m venv env
source env/bin/activate
pip install -r python-requirements.txt
which fusesoc
fusesoc --version # should return 0.1
```

Kreijstal, 2024-05-30:
why is fusesoc 2.3 not supported, why can't the changes be upstreamed?

Kreijstal, 2024-05-30:
I just saw the upstream changes of fusesoc, and the changes are ... linting?
https://github.com/lowRISC/fusesoc/