lowRISC/ibex #2140: Replacement of Xcelium

AhmedAmrAbdellatif1, 2024-02-11:
I don't have Xcelium and i can't get it, so how can i replace it and use Questa or Verilator instead?

## My Environment

 Currently Ibex, but i want to use it on my design later

**EDA tool and version:**
 Questa

**Operating system:**
 Ubuntu Linux 22

**Version of the Ibex source code:**
 Latest

rswarbrick, 2024-02-12:
Hi there.

I'd probably suggest trying to use Verilator as an initial step. The best starting point is the "simple system" (see [here](https://github.com/lowRISC/ibex/blob/master/examples/simple_system/README.md) for a description).

Our Questa support is a bit patchy, mainly because the main Ibex dev team don't have Questa licences. We'd definitely appreciate improvements, but it probably isn't the easiest place to start.

closed 2024-02-12 by rswarbrick

AhmedAmrAbdellatif1, 2024-02-12:
Hello @rswarbrick,
Sorry i forgot to mention that i need to test the integration of Ibex Core with RISCV-DV to study it, as i want to integrate my design files of RV64IMC architecture with RISCV-DV. 
Thus, i'm not able to do so because of the absence of Xcelium.

rswarbrick, 2024-02-12:
If you want to use UVM testing, you'll need a commercial EDA tool at the moment. In that case, I guess you will have to get things working with Questa. Please file bugs if you find dodgy bits!

AhmedAmrAbdellatif1, 2024-02-12:
```
juba@juba:~/ibex/dv/uvm/core_ibex$ make
Build metadata already exists, not recreating from scratch.
Building RTL testbench
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 11
make: *** [Makefile:66: run] Error 2
```