lowRISC/ibex #2175: [dv] Implement spurious memory response test

GregAC, 2024-06-07:
Secure Ibex configs explicitly check that responses received from memory are expected and ignores them if they are not.

Write a test that random injects spurious memory responses. This will be implemented similarly to the existing `riscv_mem_error_test`/`riscv_mem_intg_error_test` tests but will require new memory agent functionality to generate the spurious responses.

No new checkers are required, if the functionality is working correctly the spurious responses will have no effect and execution will continue as normal.

GregAC, 2024-07-02:
PR here: https://github.com/lowRISC/ibex/pull/2185

vogelpi, 2024-07-05:
I am closing this issue as the PR implementing this test has been merged.

closed 2024-07-05 by vogelpi