lowRISC/ibex #2180: `ecall` are counted as faults for `double_fault_seen_o`

zhangarino, 2024-06-27:
It seems like this is working as intended as all synchronous exceptions are considered a fault, including ones triggered by `ecall` (based on https://ibex-core.readthedocs.io/en/latest/03_reference/exception_interrupts.html#double-fault-detection). I was wondering however if it would be possible to exclude `ecall` from the fault definition as these exceptions are intentionally triggered for the most part.

This would be useful for RTOSes on an ibex core with `mseccfg.MMWP=1` for added memory protection. AFAIK a lot of M-mode execution in this scenario will take place after an `ecall`, so faults occuring in M-mode are more difficult to handle/debug as they will now lockup the CPU instead of causing it to go into any SW-defined exception handler.
## Observed Behavior

`ecall` in to M-mode, then trigger another synchronous exception (invalid memory-access, another ecall, etc...).

CPU lockup after an `ecall` from U-mode to M-mode, and then a synchronous exception in M-mode.

## Expected Behavior

CPU execution goes to trap table instead.

## Steps to reproduce the issue

`ecall` in to M-mode, then trigger another synchronous exception (invalid memory-access, another ecall, etc...).

GregAC, 2024-07-04:
Thanks for the report @zhangarino. I think what you suggest is reasonable though please need the `double_fault_seen_o` output is only a monitoring mechanism. It's up to the integrator how their system responds to it. As such an `ecall` followed by another synchronous exception will not lock up the core. Though if you have observed different behaviour please provide more details on the scenario.

GregAC, 2024-07-04:
Note there is a simple workaround here, I need to double check but from an RTL examination the `sync_exc_seen` field of the `cpuctrlsts` CSR is writeable by software. So if that field is cleared early on by the `ecall` handler you would not get a double fault signalled should a synchronous exception occur inside the `ecall` handler.

zhangarino, 2024-07-08:
@GregAC - Ah did not see that, thanks for the pointer! Cleared that value in my `ecall` handler on an Earlgrey chip and it accomplishes pretty much what I was looking for (getting into the exception handler reliably in M-mode)

zhangarino, 2024-07-08:
I think the "lockup" I was seeing was probably something related to the alert handler - closing this issue as the chip behavior is WAI and there's a good software workaround (modifying `cpuctrlsts`) to deal with edge cases that SW may want to exclude.

closed 2024-07-08 by zhangarino