lowRISC/ibex #2186: [rtl] Behaviour of debug single stepping with dummy instruction insertion may be surprising

GregAC, 2024-07-04:
I have observed a scenario where Ibex is single stepping whilst dummy instruction insertion is enabled. When a dummy instruction in inserted just as Ibex returns from debug mode to single step the next instruction it single steps the dummy instruction instead. The `depc` is set correctly (i.e. it will rerun the instruction that should have been single stepped after a `dret`) but it could cause confusion for a debugger user and might break the debugger itself (that may reasonably assume a single step will always occur) but the instruction won't be incorrectly skipped.

This results in a mismatch with the cosim as it doesn't observe the dummy instruction on the RVFI interface so it sees the debug vector being re-entered without the single stepped instruction being executed.

To reproduce run:

`make TEST=riscv_debug_single_step_test SEED=6250 ITERATIONS=1 WAVES=1` against commit 3384bf4c421781a2f5129307b912ac586c784977

There's three options I see here:

1. Declare this behaviour is reasonable
2. Force dummy instruction insertion to be disabled when running in single step
3. Alter the controller behaviour so we'll execute the dummy instruction and then single step

3. is the most complex, 2. should be pretty simple though does require an RTL change. 1 seems reasonable for OpenTitan (so no RTL fix needed for earlgrey)

For any of these options we should look at DV in this area to ensure this scenario is properly covered.

GregAC, 2024-07-04:
@vogelpi @andreaskurth please note and in particular give your views on whether you're happy to leave this behaviour in earlgrey.

GregAC, 2024-07-04:
Note there is a simple workaround here, I need to double check but from an RTL examination the `sync_exc_seen` field of the `cpuctrlsts` CSR is writeable by software. So if that field is cleared early on by the `ecall` handler you would not get a double fault signalled should a synchronous exception occur inside the `ecall` handler.

andreaskurth, 2024-07-04:
I think option 1 is reasonable for this release since this doesn't seem to be a severe problem, but in a future release we may want to implement option 3.

vogelpi, 2024-07-04:
Thanks for creating the issue and describing the behavior @GregAC . I share @andreaskurth 's view that this behavior is acceptable for Earlgrey. When debugging, user can also disable dummy instruction insertion but it's good to know that if it stays enabled, single stepping isn't completely broken or similar. For a future release, I would favor Option 3.

closed 2026-10-16 by andreaskurth