lowRISC/ibex #2187: [dv] Deep dive regression triage

GregAC, 2024-07-04:
# Overview

A detailed investigation of all failures observed in a local regression run of commit 1449ed5ea8f3d03127f7c3624e635e132830367d running the `opentitan` configuration was done.  Every observed failure was looked at in full (as opposed to just assuming the same signature means the same failure).  A second regression was run against 3384bf4c421781a2f5129307b912ac586c784977 which included some fixes for issues found from the first regression.  The regression log files (regr-first.log, regr-second.log) and regression HTML summaries (report-first.html, report-second.html) is attached to this issue.

This investigation has found one RTL issue but it is of low concern (and could be argued it doesn't need fixing as the behaviour is already reasonable see: #2186).  Otherwise all failures are related to environment/testbench issues and raise no concerns around the RTL.

This report first lists signatures seen and an explanation for each then it lists specific test failures seen and which explanation applies to each test failure.  For some of the signatures as there were either a significant number of them (which would have taken a long time to fully investigate) or there was some concern they could be hiding a bug fixes were made to address them.  The regression was re-run with these fixes applied to ensure the specific tests being addressed had been fixed.  Any failures remaining for the specific tests were then investigated in detail once again.  The full regression log was also reviewed for the other tests but just to look at signatures to check there was nothing obviously new, there was no detailed investigation.  All of the details of this (signatures fixed, PRs for the fixes and specific tests that got re-investigated) are given below.

# First Regression

Run against commit 1449ed5ea8f3d03127f7c3624e635e132830367d.  See the attached `regr-first.log` and `report-first.html` for the detailed regression report.

## Failure explanations and associated signatures

This provides explanations for every failure seen in the first regression and an example signature that was seen with that failure.

1. `[E] 107: UVM_FATAL /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/tests/core_ibex_base_test.sv(326) @ 32002046: uvm_test_top [uvm_test_top] Test failed due to wall-clock timeout. [1800s]`

This is a issue with RISC-V DV test generation.  Either the test has entered an infinite loop and will never terminate or will eventually terminate but it is taking too long.  In both scenarios the logs have been examined and there has been no lock-up in the design the Ibex core continues to execute instructions in a timely way up to the point of test failure with cosim matching in all cases.

2. `[E] 111: UVM_FATAL /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/tests/core_ibex_test_lib.sv(843) @ 87608: reporter [uvm_test_top] Check failed dut_vif.dut_cb.priv_mode == mode (0 [0x0] vs 3 [0x3]) Incorrect privilege mode`

This is a testbench issue related to the old checkers that are still present in some tests.  These checks are incorrectly predicting a privilege mode change.  The log of the failing test has been checked and the Ibex core is matching with the co-simulation up to the point of test failure and there are no other errors reported.

3. `[E] 118: UVM_FATAL /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv(168) @ 121679: uvm_test_top.env.cosim_agent.scoreboard [uvm_test_top.env.cosim_agent.scoreboard] Cosim mismatch Register write data mismatch to x6 DUT: 8001286d expected: 0`

This is an issue with co-simulation not correctly predicting core behaviour.  The core itself is behaving reasonably but it is a complex edge case that isn't yet correctly modelled.  We see a load response with a data integrity error that indicates a bus error.  This results in a precise exception for the bus error that is immediately interrupted by the internal interrupt for the integrity error.  After the internal interrupt handler is done we continue with the first instruction of the exception handler for the bus error (with mepc being restored correctly due to the recoverable NMI mechanism).  The mismatch happens because we write the mtval with the faulting address for the internal interrupt but this write is not carried over to spike.  So spike executes the internal interrupt handler (with no prior mtval write) which doesn't read the mtval value and then executes the exceptoion handler which does read the mtval this then mismatches.  To fix this the mtval write that occurs due the internal interrupt needs to be carried over to the co-simulation environment.

4. `[E] 162: UVM_FATAL /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv(168) @ 3437882: uvm_test_top.env.cosim_agent.scoreboard [uvm_test_top.env.cosim_agent.scoreboard] Cosim mismatch DUT didn't write to register x8, but a write was expected`

This is a mismatch between the memory model used by the co-simulation and the memory model in the UVM environment.  It is also an issue with the RISC-DV test generation (it jumps to an area of memory that hasn't been initialized).  It relates to the differing handling of unitialized memory between co-simulation and UVM.  Earlier in the test a few bytes are written to a previously unitialized area of memory.  The test then jumps to an address one byte before these written bytes.  In the co-simulation the first (unitialized) byte reads as 0x0 and the second (and following bytes) are read as the values they were previously written.  In the UVM environment it sees an instruction fetch involving an unitialized memory byte and forces a value of 0x0 for all fetched bytes.  This results in the co-simulation and Ibex core fetching different instructions from the same address and a mis-match occurs.  The behaviour of the two memory models should be unified.

5. `[FAILED]: Processing the ibex trace failed: Logfile /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/out/run/tests/riscv_pc_intg_test.14611/trace_core_00000000.log not found`

Occurrs in the riscv_pc_intg_test test.  This is an issue in the regression reporting setup.  Due to the way this test works it doesn't generate a trace file in some cases (which is flagged as an error in rethe report) but there is no failure.  The test tries to inject a corrupt PC very early in the test before execution has begun.  The corrupt PC does not trigger any alert as this particular hardening mechanism only checks anything when the Ibex core is doing straight-line execution (i.e. in particular the PC integrity check isn't carried out for branches/jumps).  Before execution has started no check is occurring so no alert is triggered, which is expected behaviour.

6. `[E] 116: UVM_FATAL /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv(168) @ 148168: uvm_test_top.env.cosim_agent.scoreboard [uvm_test_top.env.cosim_agent.scoreboard] Cosim mismatch Synchronous trap was expected at ISS PC: 80000000 but the DUT didn't report one at PC 800029ee`

An access is made to the address region spike reserves for debug memory.  This causes a fault in spike as it generates a bus error but not in Ibex.  The spike debug module is not present in the co-simulation environment so there's no need for the bus error here.  The cause of this fault has been removed (see https://github.com/lowRISC/riscv-isa-sim/commit/1f89c527c40da4d9c7984e1ae4990cdf18198276).

7. `[FAILURE] Simulation process killed due to timeout [1860s]`.
An infrastructure issue, when rerunning the test it passed successfully.

8. `131: xmsim: *SE,EILLCT: (File: ./fcov/core_ibex_fcov_if.sv, Line: 762):(Time: 730828 NS + 4) Illegal cross tuple (<0, 1, InstrCategoryFetchError, IdStallTypeMem, 0, 0, 0>) occurred corresponding to illegal cross bin (mem_stall_illegal) of cross (core_ibex_tb_top.dut.u_ibex_top.u_ibex_core.u_fcov_bind.uarch_cg_inst@11461_629.exception_stall_instr_cross).
[E] 132: xmsim: *SE,EILLCT: (File: ./fcov/core_ibex_fcov_if.sv, Line: 762):(Time: 730868 NS + 4) Illegal cross tuple (<0, 1, InstrCategoryFetchError, IdStallTypeMem, 0, 0, 0>) occurred corresponding to illegal cross bin (mem_stall_illegal) of cross (core_ibex_tb_top.dut.u_ibex_top.gen_lockstep.u_ibex_lockstep.u_shadow_core.u_fcov_bind.uarch_cg_inst@11478_1.exception_stall_instr_cross).`

[... cut ...]
 - riscv_assorted_traps_interrupts_debug_test.6257 - Signature 1
 - riscv_assorted_traps_interrupts_debug_test.6258 - Signature 1
 - riscv_pmp_basic_test.6255 - Signature 1
 - riscv_pmp_basic_test.6271 - Signature 1
 - riscv_pmp_basic_test.6288 - Signature 1
 - riscv_pmp_out_of_bounds_test.6258 - Signature 1
 - riscv_pmp_out_of_bounds_test.6261 - Signature 1
 - riscv_pmp_out_of_bounds_test.6272 - Signature 1
 - riscv_pmp_out_of_bounds_test.6282 - Signature 1
 - riscv_pmp_full_random_test.6473 - Signature 3

GregAC, 2024-07-04:
Unfortunately I cannot attach html files to a github issue so it's just the regression logs available.

vogelpi, 2024-07-05:
Thanks for your analysis and the report @GregAC . This all makes sense to me and I agree that the identified possible RTL bug isn't critical.

I compared the latest nightly regression report against your analysis and the results match. I observed one new signature:
```
riscv_debug_ebreakmu_test.11102
-------------------------------
binary:          test.bin
rtl_log:         rtl_sim.log
rtl_trace:       trace_core_00000000.log
iss_cosim_trace: spike_cosim_trace_core_00000000.log

[FAILED]: error seen in 'rtl_sim.log'
---------------*LOG-EXTRACT*----------------
    105: UVM_INFO /home/runner/_work/lowrisc-private-ci/lowrisc-private-ci/ibex/dv/uvm/core_ibex/tests/core_ibex_base_test.sv(326) @ 2025: uvm_test_top [uvm_test_top] Test wall-clock timeout is set to : 1800s
    106: UVM_INFO /home/runner/_work/lowrisc-private-ci/lowrisc-private-ci/ibex/dv/uvm/core_ibex/common/ibex_mem_intf_agent/ibex_mem_intf_response_seq_lib.sv(44) @ 2025: uvm_test_top.env.instr_if_response_agent.sequencer@@instr_intf_seq [uvm_test_top.env.instr_if_response_agent.sequencer.instr_intf_seq] is_dmem_seq: 0x0
    107: UVM_INFO /home/runner/_work/lowrisc-private-ci/lowrisc-private-ci/ibex/dv/uvm/core_ibex/common/ibex_mem_intf_agent/ibex_mem_intf_response_seq_lib.sv(44) @ 2025: uvm_test_top.env.data_if_response_agent.sequencer@@data_intf_seq [uvm_test_top.env.data_if_response_agent.sequencer.data_intf_seq] is_dmem_seq: 0x1
    108: UVM_INFO /home/runner/_work/lowrisc-private-ci/lowrisc-private-ci/ibex/dv/uvm/core_ibex/tests/core_ibex_new_seq_lib.sv(65) @ 2025: reporter@@fetch_enable_seq_h [fetch_enable_seq_h] Running the "InfiniteRuns" schedule for stimulus generation
    109: core_ibex_tb_top.dut.u_ibex_tracer.unmblk2.unmblk1: Writing execution trace to /home/runner/_work/lowrisc-private-ci/lowrisc-private-ci/ibex/dv/uvm/core_ibex/out/run/tests/riscv_debug_ebreakmu_test.11102/trace_core_00000000.log
[E] 110: UVM_FATAL /home/runner/_work/lowrisc-private-ci/lowrisc-private-ci/ibex/dv/uvm/core_ibex/tests/core_ibex_test_lib.sv(1740) @ 33765: uvm_test_top [uvm_test_top] EBreak seen whilst doing initial debug initialization, KNOWN FAILURE SEE https://github.com/lowRISC/ibex/issues/1313
    111: 
    112: --- RISC-V UVM TEST FAILED ---
    113: 
    114: UVM_INFO /nas/lowrisc/tools/cadence/xcelium/21.09-s006/tools/methodology/UVM/CDNS-1.2/sv/src/base/uvm_report_catcher.svh(705) @ 33765: reporter [UVM/REPORT/CATCHER]
--------------------------------------------
```
It looks like an old and previously fixed testbench / co-simulation issue shows up again. In any case, this issue isn't a show stopper either.

GregAC, 2024-08-23:
Some more signatures investigated whilst looking at other regressions. I won't give details of precise commits/regressions etc as this is just useful information for later triages rather than a particular point in time investigation the above posts were:

Signature:

```
riscv_assorted_traps_interrupts_debug_test.10313
------------------------------------------------
binary:          test.bin
rtl_log:         rtl_sim.log
rtl_trace:       trace_core_00000000.log
iss_cosim_trace: spike_cosim_trace_core_00000000.log

[FAILED]: error seen in 'rtl_sim.log'
---------------*LOG-EXTRACT*----------------
    294: 1261075: Illegal instruction (hart 0) at PC 0x8000ba0c: 0x00010413
    295: 1261095: Illegal instruction (hart 0) at PC 0x8000ba0c: 0x00010413
    296: UVM_INFO /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/tests/core_ibex_new_seq_lib.sv(77) @ 1264765: uvm_test_top.env.irq_agent.sequencer@@debug_new_seq_h [uvm_test_top.env.irq_agent.sequencer.debug_new_seq_h] Running 6/9
    297: 1265235: Illegal instruction (hart 0) at PC 0x8000ba0c: 0x00010413
    298: 1265275: Illegal instruction (hart 0) at PC 0x8000ba0c: 0x00010413
[E] 299: UVM_FATAL /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv(168) @ 1265285: uvm_test_top.env.cosim_agent.scoreboard [uvm_test_top.env.cosim_agent.scoreboard] Cosim mismatch Synchronous trap was expected at ISS PC: 80000004 but the DUT didn't report one at PC 8000ba08
    300: 
    301: 
    302: --- RISC-V UVM TEST FAILED ---
    303: 
--------------------------------------------

This is an issue with test generation. A run of store instructions are writing to memory just ahead of the PC (i.e. they are modifying the code we're about to execute). The first of these modified instructions executed and spike sees the new value and Ibex sees the old value. Both are reasonable behaviours (you need a FENCE.I to guarantee the changed instructions are observed by the fetch) and a mismatch results.

GregAC, 2024-08-23:
```
riscv_assorted_traps_interrupts_debug_test.10320
------------------------------------------------
binary:          test.bin
rtl_log:         rtl_sim.log
rtl_trace:       trace_core_00000000.log
iss_cosim_trace: spike_cosim_trace_core_00000000.log

[FAILED]: error seen in 'rtl_sim.log'
---------------*LOG-EXTRACT*----------------
    686: 8581206: Illegal instruction (hart 0) at PC 0x952ade74: 0x00010413
    687: 8608426: Illegal instruction (hart 0) at PC 0x952ade78: 0x00010413
    688: 8608466: Illegal instruction (hart 0) at PC 0x952ade78: 0x00010413
    689: 8619966: Illegal instruction (hart 0) at PC 0x952ade7c: 0x00010413
    690: 8620006: Illegal instruction (hart 0) at PC 0x952ade7c: 0x00010413
[E] 691: UVM_FATAL /home/greg/work/ibex_area/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv(168) @ 8650356: uvm_test_top.env.cosim_agent.scoreboard [uvm_test_top.env.cosim_agent.scoreboard] Cosim mismatch Register write data mismatch to x14 DUT: 1800 expected: 1880
    692: 
    693: 
    694: --- RISC-V UVM TEST FAILED ---
    695: 
--------------------------------------------
```

This is a co-simulation issue. In this scenario an interrupt is taken and before the first instruction of that interrupt can execute an NMI occurs. The NMI takes priority and we start executing the NMI handler. Both Ibex and spike agree in this point (in particular both reading the mcause value showing an NMI cause for the exception) but disagree on the `mstatus.mpie` value. Ibex has this as 0 and spike as 1. When the first interrupt is taken `mstatus.mpie` is set to 1 and  `mstatus.mie`to 0 (interrupts disabled for the handler and were enabled before the interrupt). In Ibex when the NMI is taken `mstatus.mpie` is set to 0 (the value of `mstatus.mie` when the NMI is taken) but spike does not do this (which is incorrect here) and there is a mismatch.