lowRISC/ibex #2196: ibex build break on older verilator version

akashkollipara, 2024-07-25:
## Observed Behavior

I am trying to build ibex using verilator installed on ubuntu 22.04, verilator version: 4.038 2020-07-11. The build breaks with error point to RV32E is not defines.

## Expected Behavior

It should be working. Many still use ubuntu 22.04 as their primary workstation. Building verialator on workstation is a bit tedious and has a lots of dependencies. So I would prefer installing it with `apt`.

## Steps to reproduce the issue

Just follow the instruction on verilator page to install on ubuntu 22.04 and try building ibex core.

## My Environment

**EDA tool and version:**

Verilator: 4.038 2020-07-11

**Operating system:**

Ubuntu 22.04.4 LTS (with latest update)

**Version of the Ibex source code:**

Using tip of github repo.

rswarbrick, 2024-07-25:
This isn't something that happens locally for us. It might be a tooling issue but I'd be reasonably surprised. Could you give the steps to generate the error and the error message that comes out?

akashkollipara, 2024-07-26:
Please use ubuntu 22.04.4 LTS
use this link to clone the repo: https://github.com/VisorFolks/cyanforge
then follow as per the README. This repo provides a build wrapper to generate ibex core.

The same steps when executed on ubuntu 24.04 builds perfectly fine.

the error I see is:
```
../src/lowrisc_ibex_ibex_simple_system_core_0/ibex_simple_system.cc: In member function ‘std::string SimpleSystem::GetIsaString() const’:
../src/lowrisc_ibex_ibex_simple_system_core_0/ibex_simple_system.cc:40:46: error: ‘class Vibex_simple_system_ibex_simple_system’ has no member named ‘RV32E’; did you mean ‘RV32M’?
   40 |   std::string base = top.ibex_simple_system->RV32E ? "rv32e" : "rv32i";
      |                                              ^~~~~
      |                                              RV32M
../src/lowrisc_ibex_ibex_simple_system_core_0/ibex_simple_system.cc:43:36: warning: enum constant in boolean context [-Wint-in-bool-context]
   43 |   if (top.ibex_simple_system->RV32M)
      |                                    ^

```

rswarbrick, 2024-07-26:
How strange! As you describe, this is going to be an old Verilator bug. The version you're using is from four years ago, so I'm not sure there's much value in papering over things here. I just checked in a Nix environment and things seem to work when building with Verilator 4.210 (released in July 2021).

If you're using Ubuntu 22.04, you might need to pull in a newer version of Verilator.

closed 2024-07-26 by rswarbrick