lowRISC/ibex #2199: Ibex simple system cannot simulate with timing flag in Verilator

kolappanc, 2024-08-06:
## Observed Behavior

We synthesized the Ibex core using Yosys and when it is simulated with --timing flag in verilator (in .core file), then the simulation fails throwing an error "Active region did not converge".  Can you please shed some light on this and advise me how to debug this issue?
## Expected Behavior

## Steps to reproduce the issue

* With the synthsized netlist, the Ibex simple system "Hello World" program is run with the -no-timing flag and it runs perfectly
* With the synthesized netlist, the same program with -timing flag throws error "Active region did not converge"
## My Environment

**EDA tool and version:**

Verilator version :- 5.018 and 5.026
**Operating system:**

Cent OS 7
**Version of the Ibex source code:**

GregAC, 2024-08-06:
So to confirm you're trying to run a gate-level simulation of Ibex within the simple system environment using Verilator?

The line referred to in the error: https://github.com/lowRISC/ibex/blob/3937e484da24a09ae97820b2defe9a85fd7cea4e/examples/simple_system/rtl/ibex_simple_system.sv#L35

Is just the top-level reset input so this doesn't tell you much (other than whatever is causing the problem is likely related to reset logic)

I'd note the simple system C++ top-level was written for older versions of Verilator and may not work correct with Verilator v5 with timing enabled, so that could be the problem. Otherwise I suspect this isn't an inherent issue in the Ibex RTL but something to do with the netlist you have or the cell library. It could be caused by a synthesis issue, a bug in the cell library or a bug in Verilator.

kolappanc, 2024-08-07:
The line number 35 for me points to the top level module 
[image]
Since I cannot debug with this method, I will try to debug the issue using a mixed multi-modal simulation of RTL and Gate level to pinpoint the issue. Please recommend other ways to debug.