lowRISC/ibex #2202: [rtl] Declare Zkt - Scalar crypto data independent execution latency compliance

GregAC, 2024-08-09:
The Zkt extension is compiled with where a certain subset of instructions have guaranteed data independent latency. Ibex can likely meet this. We should review the requirements in detail and declare compliance.

KatCe, 2025-05-05:
Hi @marnovandermaas . 
To rerun the verification, we need three main steps: 
1. Instrument the design with [CellIFT](https://github.com/comsec-group/cellift-meta). 
2. Create the verification setup with the help of 𝜇CFI's scripts.
3. Run the verification using a model checker.
Our git [repo](https://github.com/comsec-group/mucfi/blob/main/verification/common/formal/scripts/launchers/verif_cadence.tcl) contains template scripts for verifying with Cadence JasperGold, but I left out the proprietary commands. 

We usually collect all sources in one large file and generate the CellIFT instrumentation. 
What is the best way to obtain a list of all sources for a specific Ibex configuration?

marnovandermaas, 2025-05-05:
Thanks @KatCe, I understand that some proprietary commands are left out. Would you first be able to confirm that your script works on the latest main?

We use FuseSoC as a build system for Ibex. For just the core you can use the following [FuseSoC core file](https://github.com/lowRISC/ibex/blob/master/ibex_core.core). This should pull in all the needed source files. If you're needing to update a manual list of source files, you should be able to reference this [file list](https://github.com/lowRISC/ibex/blob/master/dv/uvm/core_ibex/ibex_dv.f).