lowRISC/ibex #2203: Problem with ibex_alu.sv when synthesis with genus cadence

cern143, 2024-08-10:
## Observed Behavior

Hi, I am new to both cadence and genus and got those weird error when genus read ibex_alu.sv. Any help is highly appreciated 
[genus.log](https://github.com/user-attachments/files/16570136/genus.log)

## Expected Behavior

## Steps to reproduce the issue

## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

rswarbrick, 2024-08-11:
The error lines in the attached log file start with:
```
  parameter ibex_pkg::rv32b_e RV32B = ibex_pkg::RV32BNone
                            |
Error   : Parsing error. [VLOGPT-1] [read_hdl]
        : Expecting identifier in file '/home/usr14/ibex/rtl/ibex_alu.sv' on line 10, column 29.
        : Invalid Verilog syntax is parsed, or unsupported Verilog syntax is encountered.
  parameter ibex_pkg::rv32b_e RV32B = ibex_pkg::RV32BNone
```

I think this is probably saying that Genus hasn't actually imported the package file before `ibex_alu.sv`. The list of files comes from a TCL file that I think you probably wrote (`genus_script_parallel_ibex.tcl`). My guess is that this needs fixing.

Adding support for Genus would be neat (but I can't help: I don't have a licence!). I'm closing this issue because I don't think there's enough information for anyone to help make progress. If you have a specific issue that we can help with, please re-open or open a new (more specific) issue.

closed 2024-08-11 by rswarbrick

Redamancy-520, 2024-11-26:
Hi！

I'm having the exact same problem with the Genus synthesis ibex_simple_system, and I'd like to ask how to import the ibex package in Genus before synthesizing. Or where I can find a package file to import. 

Thank you for your help!