lowRISC/ibex #2208: Why did you use kGE to estimate area and not the usual units of measurement

thinhpc1110, 2024-08-25:
I don't understand that why and how did you use kGE to estimate the area and not the usual units of measurement. 
can you teach me how the EDA can convert from the micro met to kGE

rswarbrick, 2024-08-28:
For the "usual units of measurement", I assume you mean some sort of area (micrometers squared, or similar). The problem is that this value scales with the process node, for rather obvious reasons!

Fortunately, it turns out that "the number of gates" is less dependent on the process node that we use, so is a more stable area estimate. There's a (very bare-bones) note about it at [Wikipedia](https://en.wikipedia.org/wiki/Gate_equivalent).

closed 2024-08-28 by rswarbrick

thinhpc1110, 2024-08-28:
[image]
I synthesized by cadence genus like above, so i don't know that is true or false.