lowRISC/ibex #2215: oh_raddr_a_err signal generated based on raddr_b_int

glaserf, 2024-09-17:
## Observed Behavior
The onehot read address error signal for read port **a** is generated based on the read address for read port **b**.

https://github.com/lowRISC/ibex/blob/53888bcdf4ca3c07e5e715fb6386cb4cc643a61b/rtl/ibex_register_file_latch.sv#L110-L125

## Expected Behavior
The onehot read address error signal for read port **a** should be generated based on the read address for read port **a**, unless I'm missing something.

## Version of the Ibex source code
[53888bc](https://github.com/lowRISC/ibex/commit/53888bcdf4ca3c07e5e715fb6386cb4cc643a61b) (part of [master](https://github.com/lowRISC/ibex/tree/master))

rswarbrick, 2024-09-17:
Thanks for this report, and it looks like there was a typo in 35bbdb7be. @nasahlpa: Have I understood correctly and do we just need to change a character?

GregAC, 2024-09-17:
Well spotted @glaserf!

Looks like this only effects the latch implementation of the register file, which is why I think we didn't find it previously (as we simulate with the FF based register file). We should consider refactoring here so the security hardening is factored out into some common module that works equally against the FF/Latch/FPGA implementation options.

glaserf, 2024-09-17:
> Looks like this only effects the latch implementation of the register file, which is why I think we didn't find it previously (as we simulate with the FF based register file).

This is what I supposed - I stumbled upon it as this file caused a conflict during merging to an internal development branch were we added some test structures for the non-secure Ibex variant.

nasahlpa, 2024-09-17:
Nicely spotted @glaserf! Indeed, this is a typo introduced with the recent changes in the RF hardening.

Fixed in #2216.

closed 2024-11-18 by nasahlpa