lowRISC/ibex #2222: Ask a question related to Ibex

kalabYibeltal, 2024-11-12:
I tried running Ibex small configuration with different workload by turning the Icache parameter on and off (assigning it 0 and 1) this is what I got in the Simulation statistics

Without Cache

Simulation statistics
=====================
Executed cycles:  167525
Wallclock time:   8.708 s
Simulation speed: 19238.1 cycles/s (19.2381 kHz)
Trace file size:  5758073 B

With cache

Simulation statistics
=====================
Executed cycles:  167525
Wallclock time:   14.255 s
Simulation speed: 11752 cycles/s (11.752 kHz)
Trace file size:  7273683 B

I tested it with different workloads and different configurations available in ibex. it seems like the Executed cycles should be smaller if a cache exists. On top of that it seems the wallclock time takes longer for a configuration with cache.

I also tried decreasing the clock from 0.5  to 0.05 in ibex_simple_system.sv but still the Executed cycles do not change.

## My Environment

I am using wsl on windows and I am running ibex simple system on bare metal
I am using ubuntu Ubuntu 22.04.4 LTS

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

a-will, 2024-11-12:
Did you actually enable the cache in software? The Verilog parameter affects whether the cache is present in hardware. The default state of the CPU still leaves the cache inactive.

https://ibex-core.readthedocs.io/en/latest/03_reference/cs_registers.html#cpu-control-and-status-register-cpuctrlsts

GregAC, 2024-11-22:
Enabling the icache requires a specific bit to be set by software in the `CPUCTRLSTS` CSR. We have a function that can enable/disable it:

https://github.com/lowRISC/ibex/blob/84232a5bfa8b020cd05718b2ae21d8584c942df8/examples/sw/simple_system/common/simple_system_common.h#L99-L114

Also note that in simple system the memory is dual ported with single cycle response (i.e. the most ideal possible memory from a performance perspective) and the icache won't actually have an effect. Indeed you may observe a small decrease in performance. The icache is useful in scenarios where a fetch from instruction memory can take multiple cycles (e.g. if you're fetching from flash).