lowRISC/ibex #2244: How reliable is the reference model of cosim?

ha0lyu, 2025-01-12:
I installed the co-simulation system, followed this [tutorial](https://ibex-core.readthedocs.io/en/latest/03_reference/cosim.html#setup-and-usage) to install cosim and tested it on ibex. One of the test cases shows unexpected behavior:
```
FAILURE: Co-simulation mismatch at time                 1302
Register write data mismatch to x5 DUT: 2 expected: 1
[1302] %Error: ibex_simple_system_cosim_checker.sv:68: Assertion failed in TOP.ibex_simple_system.u_ibex_simple_system_cosim_checker_bind: Co-simulation mismatch seen
Received stop request, shutting down simulation
```
I would like to ask, how reliable is the reference model of cosim? I am unsure whether this issue is necessarily a problem with ibex because the reference model version is [ibex-cosim-v0.5](https://github.com/lowRISC/riscv-isa-sim/releases/tag/ibex-cosim-v0.5).

rswarbrick, 2025-01-12:
Thanks the question. This looks a lot like it points at a bug (probably in our cosimulation environment). Can you give us step by step instructions about how to reproduce it at our end? (What command, what seed etc.)

ha0lyu, 2025-01-12:
Thanks for your reply.
1. Clone ibex, version: 8f4c75c5e
2. Follow the [Quick Build and Run Instructions](https://ibex-core.readthedocs.io/en/latest/03_reference/cosim.html#quick-build-and-run-instructions), to build and install the co-simulator
3. build simulator in ibex dir: `fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system_cosim  $(util/ibex_config.py opentitan fusesoc_opts)`
4. run cosim: `build/lowrisc_ibex_ibex_simple_system_cosim_0/sim-verilator/Vibex_simple_system -t . --meminit=ram,./examples/sw/simple_system/hello_test.elf -c 3000`.

Here is the test code:
```
  asm volatile("li t0, 0x00FF208");
  asm volatile("jalr x0, t0, 0");
```
Add the code to `examples/sw/simple_system/hello_test.c` and make, then run ibex.

In fact, this test code will jump to execute at `pc=0x000FF208`. However, the `link.ld` file defines the RAM range to start from `0x00100000`, and this `pc` is outside the RAM range. Therefore, the reference model, Spike, throws an **Instruction Access Fault** trap, while Ibex retrieves the value and decodes it, leading to an **Illegal Instruction** trap.

[image]