lowRISC/ibex #2248: pip3 install -U -r python-requirements.txt fails on pyboolector

epsilon537, 2025-01-20:
"pip3 install -U -r python-requirements.txt" fails on the installation of the pyboolector package.

This is a known issue: https://github.com/Boolector/boolector/issues/224

It won't be fixed becasue pyboolector is no longer maintained.

## My Environment

Arch Linux
Python version 3.13.1
Pip version 24.3.1

rswarbrick, 2025-01-21:
Thank you for the bug report. We think this is linked to the Python version that you are using. Running this locally seems to work (tried with Python 3.10 and Python 3.12.8).

The dependency comes through `riscv-dv` and then `pyvsc`. So I'm afraid this is probably a dependency problem in `pyvsc`. I think fixing it properly will need you to file an issue with that project.

For now, we recommend try with an older Python version (where it *should* just work!)

closed 2025-01-21 by rswarbrick

vs34, 2025-01-21:
it is not working for me also 
python3.10 --version
Python 3.10.16

i am using docker image of ubuntu 

PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian

```root@5f38c6f7ac83:/chotu/ibex# pip3 install python-requirements.txt
ERROR: Could not find a version that satisfies the requirement python-requirements.txt (from versions: none)
ERROR: No matching distribution found for python-requirements.txt

[notice] A new release of pip is available: 23.0.1 -> 24.3.1
[notice] To update, run: python3.10 -m pip install --upgrade pip
root@5f38c6f7ac83:/chotu/ibex# python3.10 -m pip install --upgrade pip
Requirement already satisfied: pip in /usr/local/lib/python3.10/dist-packages (23.0.1)
Collecting pip
  Downloading pip-24.3.1-py3-none-any.whl (1.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.8/1.8 MB 19.5 MB/s eta 0:00:00
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 23.0.1
    Uninstalling pip-23.0.1:
      Successfully uninstalled pip-23.0.1
Successfully installed pip-24.3.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
root@5f38c6f7ac83:/chotu/ibex# pip3 install python-requirements.txt
ERROR: Could not find a version that satisfies the requirement python-requirements.txt (from versions: none)
ERROR: No matching distribution found for python-requirements.txt
root@5f38c6f7ac83:/chotu/ibex# pip
pip      pip3     pip3.10
root@5f38c6f7ac83:/chotu/ibex# pip3.10 install pyvsc
Collecting pyvsc
  Using cached pyvsc-0.9.3.10985030023-py2.py3-none-any.whl.metadata (738 bytes)
Collecting pyboolector>=3.2.2 (from pyvsc)
  Using cached pyboolector-3.2.4.20240823.1.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Discarding https://files.pythonhosted.org/packages/a2/d7/96832c80e8c6f35fcfc70dc43cc7cb77ccc0e536c9f032bbe4998bd7feb8/pyboolector-3.2.4.20240823.1.tar.gz (from https://pypi.org/simple/pyboolector/): Requested pyboolector>=3.2.2 from https://files.pythonhosted.org/packages/a2/d7/96832c80e8c6f35fcfc70dc43cc7cb77ccc0e536c9f032bbe4998bd7feb8/pyboolector-3.2.4.20240823.1.tar.gz (from pyvsc) has inconsistent version: expected '3.2.4.20240823.1', but metadata has '3.2.4'
  Using cached pyboolector-3.2.3.20240822.20.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Discarding https://files.pythonhosted.org/packages/05/85/5f1dc3bd495f218df864f17a1e5bc50901b9b61a4281ca73f512f87b08b9/pyboolector-3.2.3.20240822.20.tar.gz (from https://pypi.org/simple/pyboolector/): Requested pyboolector>=3.2.2 from https://files.pythonhosted.org/packages/05/85/5f1dc3bd495f218df864f17a1e5bc50901b9b61a4281ca73f512f87b08b9/pyboolector-3.2.3.20240822.20.tar.gz (from pyvsc) has inconsistent version: expected '3.2.3.20240822.20', but metadata has '3.2.3'
  Using cached pyboolector-3.2.3.20240822.1.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Discarding https://files.pythonhosted.org/packages/e3/33/31c4d3adab5e1a0568b1970190e25e1e72f88c119787d35a035e3fb32559/pyboolector-3.2.3.20240822.1.tar.gz (from https://pypi.org/simple/pyboolector/): Requested pyboolector>=3.2.2 from https://files.pythonhosted.org/packages/e3/33/31c4d3adab5e1a0568b1970190e25e1e72f88c119787d35a035e3fb32559/pyboolector-3.2.3.20240822.1.tar.gz (from pyvsc) has inconsistent version: expected '3.2.3.20240822.1', but metadata has '3.2.3'
  Using cached PyBoolector-3.2.3.20240305.1.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Discarding https://files.pythonhosted.org/packages/27/c3/897087f7bf6ddfadc7dde21c4c38c6b0f0569e0ddff581265e4679a60ed9/PyBoolector-3.2.3.20240305.1.tar.gz (from https://pypi.org/simple/pyboolector/): Requested pyboolector>=3.2.2 from https://files.pythonhosted.org/packages/27/c3/897087f7bf6ddfadc7dde21c4c38c6b0f0569e0ddff581265e4679a60ed9/PyBoolector-3.2.3.20240305.1.tar.gz (from pyvsc) has inconsistent version: expected '3.2.3.20240305.1', but metadata has '3.2.3'
  Using cached PyBoolector-3.2.3.20240215.1.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
[... 41 lines cut ...]
          main()
        File "/usr/local/lib/python3.10/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/usr/local/lib/python3.10/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
[... cut ...]
          exec(code, locals())
        File "<string>", line 14, in <module>
      FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-87_ddfq4/CMakeLists.txt'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.