lowRISC/ibex #2253: Ask a question related to Ibex

roncedupon, 2025-02-12:
I am a beginner, just starting to learn about RISC-V and Ibex. I have installed the RISC-V GCC toolchain and Spike. When I run the simulation using the commands in the dv/uvm/core_ibex/README.md, I encounter the following error. I would like to know how to solve this. Thank you.
## My Environment
LSB Version:    core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal

**EDA tool and version:**

vcs-2018,uvm1.2,

**sim cmd**
make --keep-going IBEX_CONFIG=opentitan SIMULATOR=vcs ISS=spike ITERATIONS=1 SEED=1 TEST=riscv_arithmetic_basic_test WAVES=0 COV=0
**error description**
Error: unrecognized opcode `csrw 0x301,x10', extension `zicsr' required
Error: unrecognized opcode `csrw 0x305,x10', extension `zicsr' required
Error: unrecognized opcode `csrwi 0x747,4', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b0,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b1,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b2,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b3,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3a0,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b4,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b5,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b6,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b7,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3a1,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b8,x2', extension `zicsr' required
Error: unrecognized opcode `csrw 0x3b9,x2', extension `zicsr' required
...

marnovandermaas, 2025-02-12:
Thanks for your interest in Ibex. Are you using lowRISC's RISC-V toolchain? https://github.com/lowRISC/lowrisc-toolchains/releases

roncedupon, 2025-02-18:
Thank you very much,it's very nice of you, your method did indeed work. It seems I still have a long way to go.

closed 2025-02-18 by marnovandermaas