lowRISC/ibex #2263: having issue with PKG_CONFIG_PATH

varunmr, 2025-04-04:
Hi team,
This chandra tried running end-to end co simulation with lowrisc_ibex core and i am facing an issue as mentioned below,kindly help me out ,even i tried same with pre-compiled built having same issue.

make --keep-going IBEX_CONFIG=opentitan SIMULATOR=questa ISS=spike ITERATIONS=1 SEED=1 TEST=$TEST_NAME WAVES=0 COV=0
Build metadata already exists, not recreating from scratch.
Building RTL testbench
Traceback (most recent call last):
  File "/home/gonu.sekhar/Desktop/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 94, in _main
    subprocess.check_output(['pkg-config', '--exists'] + spike_iss_pc)
  File "/usr/lib64/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib64/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['pkg-config', '--exists', 'riscv-riscv', 'riscv-disasm', 'riscv-fdt', 'riscv-fesvr']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/gonu.sekhar/Desktop/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 149, in <module>
    sys.exit(_main())
  File "/home/gonu.sekhar/Desktop/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 96, in _main
    raise RuntimeError(
RuntimeError: Failed to find ['riscv-riscv', 'riscv-disasm', 'riscv-fdt', 'riscv-fesvr'] pkg-config packages. Did you set the PKG_CONFIG_PATH correctly?
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 1
Building randomized test generator
make[1]: *** [scripts/riscvdv.mk:61: out/metadata/instr.gen.build.stamp] Error 1
make[1]: Target 'all' not remade because of errors.
make: *** [Makefile:72: run] Error 2

## My Environment
i have followed all the steps which are mentioned in the git repo to install spike and gnu-toolchain

marnovandermaas, 2025-04-04:
Can you try the following command?
```sh
echo $PKG_CONFIG_PATH
```

For my setup I use the following exports:
```sh
export SPIKE_PATH=$_SPIKE_ROOT/bin
export PKG_CONFIG_PATH=$_SPIKE_ROOT/lib/pkgconfig:$PKG_CONFIG_PATH
```

It essentially lets the environment know where to find Spike.

marnovandermaas, 2025-04-04:
Actually there is a better explanation here: https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex#prerequisites

varunmr, 2025-04-04:
export SPIKE_ROOT=/usr/
 export SPIKE_PATH=$SPIKE_ROOT/bin
 export PKG_CONFIG_PATH=$SPIKE_ROOT/lib/pkgconfig:$PKG_CONFIG_PATH
make COSIM=1
Build metadata already exists, not recreating from scratch.
Building RTL testbench
Traceback (most recent call last):
  File "/home/gonu.sekhar/Desktop/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 94, in _main
    subprocess.check_output(['pkg-config', '--exists'] + spike_iss_pc)
  File "/usr/lib64/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib64/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['pkg-config', '--exists', 'riscv-riscv', 'riscv-disasm', 'riscv-fdt', 'riscv-fesvr']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/gonu.sekhar/Desktop/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 149, in <module>
    sys.exit(_main())
  File "/home/gonu.sekhar/Desktop/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 96, in _main
    raise RuntimeError(
RuntimeError: Failed to find ['riscv-riscv', 'riscv-disasm', 'riscv-fdt', 'riscv-fesvr'] pkg-config packages. Did you set the PKG_CONFIG_PATH correctly?
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 1
make: *** [Makefile:72: run] Error 2

still i am having same issue.

ls /usr/lib/pkgconfig/
riscv-disasm.pc  riscv-fesvr.pc  riscv-riscv.pc    
seems in the above directory 'riscv-fdt' is not available, does is cause any issue? if yes how to isolate it

marnovandermaas, 2025-04-04:
Missing riscv-fdt will cause that error. Did you build Spike from [here](https://github.com/lowRISC/riscv-isa-sim/tree/ibex_cosim?tab=readme-ov-file#build-steps)?

varunmr, 2025-04-07:
yes i have installed from the above git repo

marnovandermaas, 2025-04-07:
Where did you install Spike? As in what argument did you pass to --prefix when you configured the build? I usually try not to install to the /usr directory to avoid clashing with system installs.

Also after triggering the error can you run the following commands:
```
echo $PKG_CONFIG_PATH
which spike
spike --help
```

varunmr, 2025-04-07:
Where did you install Spike?
/tools-spike   (i have created new directory to install)
As in what argument did you pass to --prefix when you configured the build?
$RISCV  (RISCV=/tools-spike)

 echo $PKG_CONFIG_PATH
/tools-spike/lib/pkgconfig

ls /tools-spike/lib/pkgconfig
riscv-disasm.pc  riscv-fesvr.pc

which spike
/tools-spike/bin/spike

spike --help
Spike RISC-V ISA Simulator 1.1.1-dev

usage: spike [host options] <target program> [target options]
Host Options:
  -p<n>                 Simulate <n> processors [default 1]
  -m<n>                 Provide <n> MiB of target memory [default 2048]
  -m<a:m,b:n,...>       Provide memory regions of size m and n bytes
                          at base addresses a and b (with 4 KiB alignment)
  -d                    Interactive debug mode
  -g                    Track histogram of PCs
  -l                    Generate a log of execution
  -h, --help            Print this help message
  -H                    Start halted, allowing a debugger to connect
  --log=<name>          File name for option -l
  --debug-cmd=<name>    Read commands from file (use with -d)
  --isa=<name>          RISC-V ISA string [default RV64IMAFDC]
  --priv=<m|mu|msu>     RISC-V privilege modes supported [default MSU]
  --varch=<name>        RISC-V Vector uArch string [default vlen:128,elen:64]
  --pc=<address>        Override ELF entry point
  --hartids=<a,b,...>   Explicitly specify hartids, default is 0,1,...
  --ic=<S>:<W>:<B>      Instantiate a cache model with S sets,
  --dc=<S>:<W>:<B>        W ways, and B-byte blocks (with S and
  --l2=<S>:<W>:<B>        B both powers of 2).
  --device=<P,B,A>      Attach MMIO plugin device from an --extlib library
                          P -- Name of the MMIO plugin
                          B -- Base memory address of the device
                          A -- String arguments to pass to the plugin
                          This flag can be used multiple times.
                          The extlib flag for the library must come first.
  --log-cache-miss      Generate a log of cache miss
  --extension=<name>    Specify RoCC Extension
                          This flag can be used multiple times.
  --extlib=<name>       Shared library to load
                        This flag can be used multiple times.
  --rbb-port=<port>     Listen on <port> for remote bitbang connection
  --dump-dts            Print device tree string and exit
  --dtb=<path>          Use specified device tree blob [default: auto-generate]
  --disable-dtb         Don't write the device tree blob into memory
  --kernel=<path>       Load kernel flat image into memory
  --initrd=<path>       Load kernel initrd into memory
  --bootargs=<args>     Provide custom bootargs for kernel [default: console=hvc0 earlycon=sbi]
  --real-time-clint     Increment clint time at real-time rate
  --dm-progsize=<words> Progsize for the debug module [default 2]
  --dm-sba=<bits>       Debug system bus access supports up to <bits> wide accesses [default 0]
  --dm-auth             Debug module requires debugger to authenticate
  --dmi-rti=<n>         Number of Run-Test/Idle cycles required for a DMI access [default 0]
  --dm-abstract-rti=<n> Number of Run-Test/Idle cycles required for an abstract command to execute [default 0]
  --dm-no-hasel         Debug module supports hasel
  --dm-no-abstract-csr  Debug module won't support abstract to authenticate
  --dm-no-halt-groups   Debug module won't support halt groups
  --dm-no-impebreak     Debug module won't support implicit ebreak in program buffer
  --blocksz=<size>      Cache block size (B) for CMO operations(powers of 2) [default 64]

marnovandermaas, 2025-04-07:
So the content of the package config directory is where the difference is:
```sh
> ls lib/pkgconfig 
riscv-disasm.pc  riscv-fdt.pc  riscv-fesvr.pc  riscv-riscv.pc
```
Inside the repository where you built Spike can you just double check you are on the right remote and branch? When I do `git log` this is the first thing that pops up:
```
commit 39612f93837122a980395487f55b2d97d29d70c1 (HEAD -> ibex_cosim, origin/ibex_cosim)
Author: Greg Chadwick <gac@lowrisc.org>
Date:   Thu Jun 27 14:46:45 2024 +0200

    Introduce pre_val MIP concept.
    
    When Ibex is executing an instruction it is possible for MIP to change
    whilst that instruction is stalled in the ID/EX stage. This results in
    the MIP observed by a CSR instruction differing from the MIP that
    decides whether or not that instruction may be interrupted which leads
    to model mis-matches.
    
    This adds a new MIP 'pre_val' which is the MIP value used to determine
    whether or not an interrupt is taken. The existing value is the one
    observed by CSR instructions. Employing this mechanism avoids the
```

Also, to help you proceed, I've attached a tarball of my current installed version of Spike so you can use that in the meantime (built on Ubuntu 24.04):
https://drive.google.com/file/d/1FaS6_zhW3CYbNS8529b5hE3iH5h1h_1K/view?usp=sharing

varunmr, 2025-04-08:
Thanks for the support the above information helps me.
i am running rtl simulation with questa but its triggering xrun ,can you help me on this 

 make COSIM=1
Build metadata already exists, not recreating from scratch.
Building RTL testbench
[Errno 2] No such file or directory: 'xrun'
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 1
make: *** [Makefile:72: run] Error 2

marnovandermaas, 2025-04-09:
In case you've run the make file before you need to remove the `out` directory under `dv/uvm/core_ibex`. The meta-data does not get rebuilt when you change the parameters.

varunmr, 2025-04-11:
make TEST=riscv_machine_mode_rand_test ITERATIONS=1
Build metadata already exists, not recreating from scratch.
Building RTL testbench
make[1]: *** [scripts/ibex_sim.mk:42: out/metadata/tb.compile.stamp] Error 5
make: *** [Makefile:72: run] Error 2
after removing the out directory i got the above error.

can you please help to run the simulation with questa simulator

marnovandermaas, 2025-04-11:
I don't have access to the Questa simulator so won't be able to help on that front. Our simulator should work under Xcelium.