lowRISC/ibex #2264: Facing issue with ibex DUT

vinomutty, 2025-04-08:
Hi team

After the simulation i am facing the below error
  
make: *** [/Projects/marmik_project/hemashri.bhagavati/april/ibex/riscof_work/Makefile.DUT-ibex:9: TARGET1] Error 1
Flag could not be matched: isa

**STEPS FOLLOWED:**

cloned by using this: git clone **https://github.com/Abdulwadoodd/ibex**
Inside ibex again cloned complince tests:  git clone **https://github.com/riscv-non-isa/riscv-arch-test.git**

**For env setup**
riscof setup --dutname=ibex
 riscof --verbose info arch-test --clone
riscof validateyaml --config=config.ini

 
**EXPORT COMMANDS:**
export SAIL_RISCV=/tools/sail-riscv-new/c_emulator
export PATH=/tools/sail-riscv-new/c_emulator:$PATH

**RUN COMMAND:**
riscof run --config=config.ini --suite=riscv-arch-test/riscv-test-suite/rv32i_m/C  --env=riscv-arch-test/riscv-test-suite/env

vinomutty, 2025-04-11:
yes ,cloned from the main branch and setup the env. while simulating getting below errors
1) **ERROR | Flag could not be matched: isa
  ibex {OPTIONS} [filename]**

2)make: *** [/Projects/marmik_project/hemashri.bhagavati/master_ibex/ibex/riscof_work/Makefile.DUT-ibex:117: TARGET28] Error 1
    INFO | Running Tests on Reference Model.
    INFO | Initiating signature checking.
   ERROR | Signature file : /Projects/marmik_project/hemashri.bhagavati/master_ibex/ibex/riscof_work/src/cadd-01.S/dut/DUT- 
   ibex.signature does not exist

STEPS:
1)cloned by using: git clone **https://github.com/Abdulwadoodd/ibex.git**

2)inside the ibex copied **RISC-V Compliance Test** BY using below 
git clone **https://github.com/riscv-non-isa/riscv-arch-test.git**

marnovandermaas, 2025-04-14:
It looks like your steps include cloning your fork of Ibex which is 246 commits behind the upstream. Would you mind checking whether it is still and issue on mainline Ibex?

marnovandermaas, 2025-04-14:
We also have a vendored in version of our compliance test: https://github.com/lowRISC/ibex/tree/master/dv/riscv_compliance
I'm not sure how much that is tested, but it's something you can try out.

marnovandermaas, 2025-04-14:
Our main DV flow that is guaranteed to work is the co-simulation environment that you can find here:
https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex