lowRISC/ibex #2278: Store instructions fetched from non-executable memory update data_we_o, data_addr_o, and data_wdata_o

cnokes14, 2025-05-19:
## Observed Behavior

When a store instruction is read from a section of memory marked by the PMP as non-executable, data_we_o, data_addr_o, and data_wdata_o are still updated as though the instruction is valid. Additionally, no output alert or fault signals are raised, though the appropriate CSRs are updated.
[image]
[image]

## Expected Behavior

Instructions deemed invalid by the PMP should be entirely gated and have no artifacts. Additionally, I would expect an alert_minor_o to be raised, though I may be wrong about that.

## Steps to reproduce the issue

I've created [a repository with the test bench I used](https://github.com/cnokes14/Ibex-invalid-store-TB/blob/main/ibex_invalid_store_tb.sv) to verify the issue. 
To trigger this issue in your own environment:

1. Configure the PMP to have areas that should not be executable.
2. Store memory write instructions in that non-executable area.
3. Branch to the non-executable area.
4. Observe the error.

Regarding parameters, I observed this issue with BranchTargetALU enabled and disabled, with WritebackStage enabled and disabled, with ICache enabled, with BranchPredictor enabled and disabled, and with SecureIbex disabled. All other parameters (excluding PMP configuration) were left at their default values.
## My Environment

**EDA tool and version:**

Xilinx Vivado 2024.1

**Operating system:**

Windows 11

**Version of the Ibex source code:**

Commit SHA: 125445f692d5dc91892a7852112ad3633f44969f. No changes to source code.

marnovandermaas, 2025-05-20:
Thanks for working with Ibex! Can you check the value of `pmp_req_err[PMP_I]` and `pmp_req_err[PMP_I2]` for the cycle where the store instruction is fetched (which is where `pc_if` is equal to the program counter in question)? As far as I can tell this PMP request error signal turns into a fetch error being indicated from the fetch to the decode stage and the LSU request signal is gated on that signal. So as far as I can tell the RTL conforms to your expected behaviour.

cnokes14, 2025-05-20:
> I think the observation is that while the `data_req` is gated on the fetch error, it would appear that we, addr and wdata are not gated, and the expectation is that it should be 'entirely gated and have no artifacts' resultant from the failing insn.

Yes, this was my observation. Are my expectations for the gating mechanism incorrect?

marnovandermaas, 2025-05-22:
Thank you both for clarifying this issue. The [bus protocol](https://ibex-core.readthedocs.io/en/latest/03_reference/load_store_unit.html#protocol) states that anything on the address, write enable, byte enable and write data are don't care when the request is low, so I think Ibex's behaviour is fine here.

With regard to your minor alert point, these are mainly for hardware security issues like [instruction cache ECC errors](https://ibex-core.readthedocs.io/en/latest/03_reference/security.html#icache-ecc). PMP errors are handled directly by the exception handler.