lowRISC/ibex #2294: Ask a question related to Ibex alu bitcounting

ESE-2019, 2025-07-13:
What is the purpose of `bitcnt_bit_mask & ~bitcnt_mask_op`? I find it that it still works properly without `& ~bitcnt_mask_op`. Here are the rawcode in ibex_alu and my C test program. 

Rawcode:
```
    always_comb begin
      bitcnt_bit_mask = bitcnt_mask_op;
      bitcnt_bit_mask |= bitcnt_bit_mask << 1;
      bitcnt_bit_mask |= bitcnt_bit_mask << 2;
      bitcnt_bit_mask |= bitcnt_bit_mask << 4;
      bitcnt_bit_mask |= bitcnt_bit_mask << 8;
      bitcnt_bit_mask |= bitcnt_bit_mask << 16;
      bitcnt_bit_mask = ~bitcnt_bit_mask;
    end

    assign zbe_op = (operator_i == ALU_BCOMPRESS) | (operator_i == ALU_BDECOMPRESS);

    always_comb begin
      unique case (1'b1)
        zbe_op:      bitcnt_bits = operand_b_i;
        bitcnt_cz:   bitcnt_bits = bitcnt_bit_mask & ~bitcnt_mask_op; // clz / ctz
        default:     bitcnt_bits = operand_a_i; // cpop
      endcase
    end
```
C test code:

```
#include <stdio.h>
#include <stdint.h>

uint64_t check_pass_num = 0;

void check(uint32_t bitcnt_mask_op)
{
      uint32_t bitcnt_bit_mask;

      bitcnt_bit_mask = bitcnt_mask_op;
      bitcnt_bit_mask |= bitcnt_bit_mask << 1;
      bitcnt_bit_mask |= bitcnt_bit_mask << 2;
      bitcnt_bit_mask |= bitcnt_bit_mask << 4;
      bitcnt_bit_mask |= bitcnt_bit_mask << 8;
      bitcnt_bit_mask |= bitcnt_bit_mask << 16;
      bitcnt_bit_mask = ~bitcnt_bit_mask;
      if (bitcnt_bit_mask != (bitcnt_bit_mask & ~bitcnt_mask_op))
            printf("%x check fail\n", bitcnt_mask_op);
      else
            check_pass_num++;
}

int main()
{
      for (uint32_t i = 0; i < UINT32_MAX; i++)
            check(i);
      check(UINT32_MAX);
      printf("check_pass_num = %ld\n", check_pass_num);
      return 0;
}
```
Result: `check_pass_num = 4294967296`

## My Environment

**EDA tool and version:**

gcc version 13.3.0 (Ubuntu 13.3.0-6ubuntu2~24.04)
**Operating system:**

Ubuntu Linux 24.04

**Version of the Ibex source code:**

e5b8d05fea3edd00a068d41f2160fe9076090952

rswarbrick, 2025-07-14:
Tracing through the code, the `always_comb` block looks like a rather convoluted way of replicating that bit across the whole word and then inverting it. I think this line would have been equivalent (and easier to read):
```systemverilog
assign bitcnt_bit_mask = {32{~bitcnt_mask_op}};
```

At this point, the language gets a bit complicated to untangle. Because `bitcnt_bit_mask` is 32 bits wide, the binary `&` expands both operands to 32 bits (see row 3 of the table in section 11.6.1 of IEEE 1800-2017). But then row 4 of the table says that the bit-length used for `bitcnt_mask_op` is also 32. As such, the negated value is going to be `{31{1'b1}, ~bitcnt_mask_op}`. That means that the AND does nothing, as you say: for bit zero, it is `(~bitcnt_mask_op) & (~bitcnt_mask_op) = ~bitcnt_mask_op` and for the other bits, it is `(~bitcnt_mask_op) & 1'b1 = ~bitcnt_mask_op`.

How silly! Fancy making a PR to get rid of the `always_comb` and drop the AND?

rswarbrick, 2025-07-18:
Ha! That's rather a neat way to do it :-) Looks good to me.